ethz-asl/rotors_simulator #222: The rotors_simulators can not launch two or more same mavs at the same time

HitszChen, 2015-04-09:
I think the reason is that the macro property "namespace" is defined in firefly.xacro

file: firefly.xacro
//22 <robot name="firefly" xmlns:xacro="http://ros.org/wiki/xacro">
//23   
//24   <xacro:property name="namespace" value="firefly" />   <----------------HERE
//25   <xacro:property name="rotor_velocity_slowdown_sim" value="10" />
//26   <xacro:property name="use_mesh_file" value="true" />
//27   <xacro:property name="mesh_file" value="firefly.dae" />"

I suggest to change line 24 to
//<xacro:property name="namespace" value="$(arg namespace)" /> 
and define an argument "namespace" in the launch file. What do you think guys?

ffurrer, 2015-04-10:
Sounds good to me, will fix this after the release.

ffurrer, 2015-08-13:
This should have been resolve in #242

closed 2015-08-13 by ffurrer