ethz-asl/rotors_simulator #240: Process has died -import error

Annaiyan, 2015-06-11:
hello,

I'm new to ros and gazebo.Now,I'm trying to launch the firefly in gazebo and i can successfully do the catkin_make.But while launching the file the process has died due to the import error of python.
This is what it looks like

Traceback (most recent call last):
  File "/opt/ros/indigo/lib/gazebo_ros/spawn_model", line 25, in <module>
    from gazebo_ros import gazebo_interface
  File "/opt/ros/indigo/lib/python2.7/dist-packages/gazebo_ros/gazebo_interface.py", line 9, in <module>
    from gazebo_msgs.msg import *
ImportError: No module named msg

I don't know how to exactly solve this issue?can anyone help me !

Regards,
Annaiyan

ffurrer, 2015-06-11:
do you have `gazebo_msgs` in your workspace?

You can check so by e.g.
`roscd gazebo_msgs`

If this is not the case you either need to install it or it could be that you didn't source your environment properly.

Do so by `source ~/catkin_ws/devel/setup.bash` (assuming your catkin workspace is in your home folder in `catkin_ws`)

If this was not helpful tell us a bit more about your setup what OS are you using, what ROS version did you follow all the installation steps?

Annaiyan, 2015-06-11:
Thanks for kind information.

I tried your instructions ,

First ,when i give roscd gazebo_msgs,the entire path is changing from ~/Workspace/catkin/src to /opt/ros/indigo/share/gazebo_msgs.

Second,when i give source ~/catkin/devel/setup.bash it says there is no such file or directory.

I'm ros indigo and Ubuntu 14.04.2 LTS,Release:	14.04, Codename:	trusty.

Do you get any ideas?

ffurrer, 2015-08-13:
@Annaiyan did you resolve your issue?
(I think your workspace was located at `~/Workspace/catkin/` instead of `~/catkin_ws/` as assumed in my instructions.)

Annaiyan, 2015-08-17:
@ffurrer Yes,i have solved the issue finally and as you mentioned earlier,the exact problem is gazebo_msgs ! Thanks anyway for the support.

closed 2015-08-17 by Annaiyan

BakhtawarRehman, 2017-12-04:
@Annaiyan  I am having the same problem. 
Can you please explain: How did you solve the problem with gazebo_msgs? 
For me: gazebo_msgs are in the same path as /opt/ros/indigo/share/gazebo_msgs$ 

I have already spent 3 days on this thing. Can you please help??