ethz-asl/rotors_simulator #270: Sometimes the ROS subscription to command_sub_topic of a motor plugin doesn't work

ffurrer, 2015-08-19:
In the motor plugin,  [here](https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo_plugins/src/gazebo_motor_model.cpp#L111): 
```C++
command_sub_ = node_handle_->subscribe(command_sub_topic_, 1, &GazeboMotorModel::VelocityCallback, this);
```
the binding to the callback sometimes doesn't work, hence `VelocityCallback` doesn't get called, even though the message got published.
(the other motor plugins subscribing to the same topic are getting into their callbacks)

This happens at random instances. I don't know a way of reproducing it, but it happens occasionally if a lot of helicopters get spawned, as happens if running:

```
roslaunch rotors_gazebo firefly_swarm_hovering_example.launch
```

The motor plugin is running fine otherwise, as it publishes messages.

ffurrer, 2015-08-19:
@tfoote do you have an idea why this can happen, is this some bug/weird behaviour of ROS?

tfoote, 2015-08-19:
I've never heard of an issue like that. If the sleep makes a difference it might be isolateable into a test case.

ffurrer, 2015-08-19:
@tfoote thanks, I'll try to create a test tomorrow.

ffurrer, 2015-08-24:
I quickly tried to reproduce the error last week, but was not successful in https://github.com/ffurrer/test_subscribers

fmina, 2017-02-28:
@ffurrer any update on this? It is painful to get multiple MAVs to work.

ffurrer, 2017-03-01:
@fmina nope, sorry, didn't look into this anymore :(

fmina, 2017-03-01:
@ffurrer 
No worryings, I have a fix for this and will open a PR after IROS

ffurrer, 2017-03-01:
perfect 👍