ethz-asl/rotors_simulator #276: Notes on PX4 upstream merge

LorenzMeier, 2015-10-17:
@ffurrer @tumbili @devbharat this is to track what is needed to bring our changes back to upstream:

  * Bring models like the Iris into Gazebo: https://github.com/PX4/rotors_simulator/pull/3
  * Factor out all ROS parts from the core simulation: https://github.com/PX4/sitl_gazebo
  * Bring in MAVLink: https://github.com/PX4/rotors_simulator/commits/master

@devbharat We found multiple issues with the physics in the sitl_gazebo fork, which clearly hints we should focus on maintaining a single simulator.

devbharat, 2015-10-17:
I agree in that we need to maintain a single core simulator. I still do not see where the issues in physics of the sitl_gazebo were, the 'core' code is the same, and flies fine as is. See https://github.com/PX4/Firmware/issues/2956

LorenzMeier, 2016-01-02:
Yet another Gazebo plugin - I'm working on the coordination:
https://github.com/AurelienRoy/ardupilot_sitl_gazebo_plugin/issues/2

ffurrer, 2016-01-02:
@LorenzMeier thanks =)

nicolaerosia, 2016-03-12:
@LorenzMeier I can try to reduce the delta between the plugins which are found in https://github.com/PX4/sitl_gazebo and in here.

ffurrer, 2016-04-14:
@pvechersky thanks for https://github.com/ethz-asl/rotors_simulator/pull/288, just adding this here as a reference.

khancyr, 2016-06-05:
Hi,

I am working on using ardupilot stack on gazebo ( 3 are already present but none working out of the box). Should I base my work on https://github.com/PX4/sitl_gazebo/commits/master fork or https://github.com/ethz-asl/rotors_simulator . My main consern is about mav msg since you are using 2 differents syntaxes ...

squilter, 2016-06-15:
@khancyr Have you made any progress? I'm trying to get ArduCopter flying in Gazebo, but none of the plugins are working for me.

khancyr, 2016-06-28:
@squilter got many problems to have a simple solution on APM. But I can explain you how to rebase existing commit on master and make it works (althought there are not optimal..)

LorenzMeier, 2016-06-28:
If you are not limited to using Arducopter you could give this a go, which is already completely integrated and ready to go: http://dev.px4.io/simulation-gazebo.html

khancyr, 2016-06-28:
squilter find a better option for Arducopter : https://bitbucket.org/osrf/gazebo/branch/aero_default

closed 2018-12-06 by ffurrer