ethz-asl/rotors_simulator #281: adding a stereo camera in astec firefly

Annaiyan, 2016-02-11:
Hi,I would like to add a stereo camera in the astec firelfy in gazebo.But i don't know,how to use the components_snippets.xacro file correctly?
Can anyone help to me to add the stereo camera (in which file i have to amke the changes)?

ffurrer, 2016-02-12:
Hi,

There is already a model with a stereo camera mounted on the firefly, please check: https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_description/urdf/firefly_vi_sensor.gazebo

or just use this launch file: https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo/launch/mav_hovering_example_with_vi_sensor.launch

Let us know if you need further advise on how to use the snippets, in case this was not what you were looking for.

cheers

Annaiyan, 2016-02-12:
Thanks.Now i can launch the firefly with VI sensor but i can see only the left camera not the right camera.How to view the right camera images?(i can see the left camera images using image_view in ros)

ffurrer, 2016-02-12:
If you are on Linux, this happens occasionally, this has something to loading of the plugins in Gazebo that we couldn't figure out yet.
What operating system and what ROS/Gazebo versions do you use?
And can you try to launch it several times and check whether it's always failing to display the cameras?

Annaiyan, 2016-02-12:
I'm using ubuntu 14.04 LTS with indigo/gazebo 5.1.0.Yes i,tried couple of times but it only shows the left camera images.

Annaiyan, 2016-02-12:
is it possible to use the stereo_view(rosrun image_view stereo_view stereo:=/my_stereo_cam image:=image_rect_color) in your package to display both the camera?

won13y, 2016-07-05:
@ffurrer  
Thank you for the reply.
but, 

There used to be a model with a stereo camera mounted on the firefly as following link.

but following link is not available:

https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_description/urdf/firefly_vi_sensor.gazebo

pvechersky, 2016-07-05:
@won13y 

Some of the RotorS description and launch files have been made more general, hence many model-specific files have been removed. So now you can find the model with a mounted VI sensor here:

https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_description/urdf/mav_with_vi_sensor.gazebo

It receives the name of the mav as an argument. You can, for example, run the following launch file to spawn this model:

https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo/launch/mav_hovering_example_with_vi_sensor.launch

I hope this helps!