ethz-asl/rotors_simulator #282: Wrong octomap generation

jvgomez, 2016-02-11:
Hi,

I am using an adapted version of the octomap plugin (basically, I just created a package out of the necessary code from the master branch at today's state). The octomap visualized in Octoviz is not as it should, creating like stripes of fully occupied/fully empty spaces, as shown in this image:
[image]

[I am attaching the mesh. ](https://github.com/ethz-asl/rotors_simulator/files/126809/warehouse.zip). It would be great if somebody can check and tell whether they have the same problem. I've found more examples of faulty meshes and cases in which the octomap looks great.

Or if somebody can guess where the problem is I would investigate further.

helenol, 2016-02-11:
Hi Javier,
This is a known issue, especially when several objects are close together or share the same edge due to the way the the raycasting is done when generating the octomap.

If you are interested in improving the octomap generation, I'm sure that would be very appreciated by all. :)

A few things I can think of to try would be decreasing the leaf size here: https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo_plugins/src/gazebo_octomap_plugin.cpp#L177 or actually taking a deeper look into how the ray casting is done.

jvgomez, 2016-02-11:
Hi Helen,

Thanks for the info. I guess I will look into it since it is something I actually need, although I am not sure when I will have time for this.

In any case, any further information about the issue or the plugin to help the debugging is welcome.

Actually, I see that the raycasting is done manually. Before finding this plugin I was considering simulating a laser (or depth camera) into many positions and let Gazebo do the raycasting. Is there any reason not to do it that way?

helenol, 2016-02-12:
If you can get in poses to sufficiently observe all parts of the environment you care about, then yes, this is a great option. I believe the plugin was for generating big environments quickly (without manual intervention).

ffurrer, 2016-02-12:
Hi Javier,

I quickly had a look at your model. It doesn't work, as it consists of multiple objects that are intersecting with each other. And your walls have zero thickness, so our way of ray casting to find the beginning and ending of a wall doesn't work if you have another object in-between or if there is no other side of the object. You experienced this in the case of the main building where you have these vertical windows or by the side entrance where you have this little wall on one side and the stairs on the other side.

1. If you have time to fix the plugin to also handle objects with wall-thicknesses of zero, please open a pull-request. 
2. If you only want to get it to work for your model, change the wall-thickness to something greater than 0.

Hope that helps and keep us updated.
Cheers

jvgomez, 2016-02-12:
I understand that the raycasting might fail for walls with zero thickness, and I guess I was assuming that the plugin works in a different way than I expected. I will study the code further since it will be nice to make it able to deal with these kind of models since I have just got this model from the 3D warehouse and most of them are like this.

Do you have any further documentation to easily understand how the raycasting is done in this case?

ffurrer, 2016-02-12:
I think there are some comments in https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo_plugins/src/gazebo_octomap_plugin.cpp#L169 and the whole method, that should help you understand what we're doing.

Basically we're starting at different locations (with voxel spacing) in a loop (https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo_plugins/src/gazebo_octomap_plugin.cpp#L202) and then start shooting in one direction (from far away), as soon as we hit the object we assume to be inside the object and set all voxels to occupied, until we hit it again (unless we never hit it again).

You see, this is not a perfect solution, but by the time we did this, we just made sure that we deal with proper models :-S.

jvgomez, 2016-02-12:
OK, thanks for the very useful information :) I will see what I can do.

ZacharyTaylor, 2016-02-18:
Hi Javier,

I also found myself needing to handle intersections and zero thickness geometry so I have had a go at modifying the plugin to handle these cases.

It now works in 3 stages
1) Checks if a model edge exists in each cell and if it does marks it as occupied
2) Flood fills the area from the top and bottom marking all connected unknown cells as unoccupied
3) Marks all remaining unknown areas as occupied

This method has some situations where it may erroneously mark cells as filled but for my applications generally does a decent job.

It can be found in the feature/surface_octomap branch https://github.com/ethz-asl/rotors_simulator/tree/feature/surface_octomap

Would appreciate any further improvements you can make, but it handles the mesh you linked.

[image]

jvgomez, 2016-02-18:
@ZacharyTaylor thank you so much! I guess that the problem that it can wrongly mark as occupied unoccupied cells is because in step 2 you are doing it only for connected unknown cells, right? What is the reason behind this?

ZacharyTaylor, 2016-02-18:
With zero thickness geometry it is impossible to know if an object is meant
to be hollow or solid.

The approach assumes that all completely enclosed areas are densely
occupied. However you can alter step 3 pretty easily to set these areas as
unoccupied or to leave them as unknown if this would better suit your
situation.

jvgomez, 2016-03-10:
Sorry for the late reply. I saw that you have actually merged the newer version. Did you find any problems? I haven't had time to test it yet.

ZacharyTaylor, 2016-03-11:
I have used it in several cases where the previous method failed and it
appears to have worked well. While I could construct an environment in
which it would fail I feel that the situations where this would occur are
pretty unusual.  Let me know if you encounter an application where it
performs poorly

ffurrer, 2016-03-17:
Please reopen, if the problem is still occurring.

closed 2016-03-17 by ffurrer

jvgomez, 2016-03-22:
So far the problem seems to be solved for me, but I have a question regarding the output and I do not think I should open another issue (tell me so in that case).

The point is that it looks like the octomap generated is centered in the bounding box center in the service call, which makes some sense. But then it is hard to align the generated octomap with the corresponding mesh for simulation as most of the times the origin of the octomap will not be its center. What is the best practice? I am considering to modify the code so that the input includes an offset or something.

How do you align the generated octomap with the mesh? Or you just put the center in 0,0,0 and then make it also scan under the ground?

Thanks!!