ethz-asl/rotors_simulator #284: Waypoint publisher not working

Annaiyan, 2016-02-24:
I have installed the latest rotors simulators package and I'm trying to launch the waypointpublisher in the launch as well as separately (rosrun rotors_gazebo waypoint_publisher).But it is not working (error:Usage: waypoint_publisher <x> <y> <z> <yaw_deg> [<delay>]).
what might be the issue/problem?

orangelynx, 2016-05-09:
Hey,

you have to specify the coordinates for the new waypoint, the error couldn't be any more self-explanatory.

Try
`rosrun rotors_gazebo waypoint_publisher 1 0 1 0 0`
to fly to the coordinates (X=1,Y=0,Z=1,Yaw=0) instantly.

However there still seems to be a problem because when I use this publisher with the mav_hovering_example.launch, it says "no subscriber available".

helenol, 2016-05-09:
Try running it in the firefly namespace: 
`rosrun rotors_gazebo waypoint_publisher 1 0 1 0 0 __ns:=firefly`
If you take a look at the mav_with_waypoint_publisher.launch file, you'll see that the node is launched in a _group_ named `"$(arg mav_name)$"` which by default resolves to `firefly`. That is why all topics, service calls, etc. are prefaced with '/firefly/my_cool_topic_name'.
Running the node with `__ns:=firefly` in the input will tell it to run in that namespace.

orangelynx, 2016-05-09:
thanks, that works! In the tutorial chapter it looks like there is a space between the two underscores of '__ns' which is why it didn't work. I also failed to realize that this option belongs to rosrun and therefore I searched for the code handling that argument in the waypoint_publisher.cpp in vain.

orangelynx, 2016-05-09:
One more comment: firefly seems to refuse to obey sometimes..

Test: Publish waypoint 0,0,5 -> firefly goes there (rather excitedly)
Publish waypoint 0,0,4 -> firefly doesn't move a bit.

otherwise I think this can be closed.

kashifzr, 2023-09-23:
I am bit late here, can any one know to pass waypoint file with waypoint_publisher_file in rotor simulator 
i did make simple waypoint file in .txt format with values of delay time x y z and yaw but the output is of running
the text inside file is like 

5 0.0 2.0 2.0 0.0
5 0.0 3.0 2.0 0.0

 rosrun rotors_gazebo waypoint_publisher_file waypoints __ns:=iris
[ INFO] [1695457763.752247514]: Started waypoint_publisher.
[ERROR] [1695457763.752657278]: Unable to open poses file: waypoints