ethz-asl/rotors_simulator #285: waypoint_publisher_file doesn't publish

Asiron, 2016-02-26:
I found a possible bug, that happens in waypoint_publisher_file.
[ https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo/src/waypoint_publisher_file.cpp](url)
It doesn't publish any message, unless a statement is executed after:
```
...
wp_pub.publish(msg);
return 0;
```
for example:
```
...
wp_pub.publish(msg);
ROS_INFO("Closing..."):
return 0;
```

Let me know if you can replicate it and I can make a pull request.

ffurrer, 2016-02-29:
I have never experienced this, @burrimi @helenol @fmina anyone?

@Asiron can you post the waypoint file you're using and the command, such that we can possibly replicate this?

cheers

helenol, 2016-02-29:
Had recent problems with publishing in one-off nodes too actually, you
should just add a ros::spin() or a ros::spinOnce() (or twice...) after.

Asiron, 2016-02-29:
@ffurrer Below is the file, we used in our lab.
```
10.0 1.0 -1.0 10.0 0.0
12.0 1.0 -2.0 9.0 0.0
10.0 2.0 -1.0 9.0 0.0
10.0 3.0 -2.0 10.0 0.0
10.0 4.0 -1.0 10.0 3.0
8.0  5.0 -1.0 9.0  2.0
15.0  7.0 -1.0 20.0 1.0
15.0  9.0  3.0 20.0 2.0
15.0  10.0 -5.0 25.0 0.0
15.0  10.0 5.0 20.0 0.0
10.0  11.0 7.0 10.0 0.0
10.0  12.0 5.0 12.0 0.0
10.0  14.0 6.0 10.0 0.0
10.0  16.0 -2.0 10.0 0.0
10.0  17.0 -4.0 10.0 0.0
```
@helenol Yea, I think that might be the issue. Altough, ROS_INFO is enough, it could be the case, that sometimes it wouldn't publish it neither. I think the program returns, before the other thread publishes the message.

Musab-Obaid, 2016-04-12:
I have experienced the same issue, and the suggested solutions worked.

Musab-Obaid, 2016-04-12:
I experienced the same issue, and the suggested solution worked.

Musab0, 2016-04-12:
I have also experienced the same issue, and the suggested fix worked.

lawenyuan, 2018-01-21:
you can add following code after publish message:
‘’‘
ros::Duration(10).sleep();
‘’‘