ethz-asl/rotors_simulator #291: Documentation of controllers and launch files

fredrikbissmarckatfoise, 2016-04-18:
Hi,
I am trying to evaluate if rotorS is a viable research tool. Installation was no problem, but now I am a bit halted on how to control flights. 

- How to advance the MAV towards a way point, i.e. in the power plant example?
-  How to control the MAV with a joy stick? My game pad (Logitech f310) gives responses, I can tilt but not fly. However a sligth touch of one button invokes a powerful thrust making the firefly going sky high out of sight very quickly.

Some pointers would be very much appreciated, I am impressed of what I've seen so far..

/Fredrik, Sweden

ToniRV, 2019-04-05:
Same for thrust sending the drone to infinity

ToniRV, 2019-04-06:
Here is what I did to get it right:
```
-    <node name="rotors_joy_interface" pkg="rotors_joy_interface" type="rotors_joy_interface" />
+    <node name="rotors_joy_interface" pkg="rotors_joy_interface" type="rotors_joy_interface">
+      <param name="max_thrust" value="30.0"/>
+      <param name="max_yaw_rate" value="10.0"/>
+      <param name="axis_thrust_" value="1"/>
+      <param name="axis_roll_"   value="3"/>
+      <param name="axis_pitch_"  value="4"/>
+      <param name="button_control_enable_"  value="0"/>
+      <param name="button_control_mode_"  value="1"/>
+      <param name="button_takeoff_"  value="2"/>
+      <param name="button_land_"  value="3"/>
+      <param name="button_yaw_left_"  value="4"/>
+      <param name="button_yaw_right_"  value="5"/>
+    </node>
```