ethz-asl/rotors_simulator #306: Missing header files

sytelus, 2016-10-05:
It looks like attitude_control.h is referencing below two files:

```
#include "rotors_control/controller_base.h"
#include "rotors_control/controller_factory.h"
```

However I can't find these two headers anywhere in repo. Are they missing? Is this branch still compilable?

pvechersky, 2016-10-06:
@sytelus,

The two header files you're referring to used to be part of the 'rotors_control' package but were removed in pull request #177 ... So I think those includes should be removed, but in the meantime the branch should still compile.

sytelus, 2016-10-06:
Thanks! I'm wondering if attitude controller is working or tested. If its still in works, I would avoid using it for now.

pvechersky, 2016-10-10:
@sytelus 

My bad, I just now saw that the attitude controller files do not actually get compiled. I haven't really dealt with the controller side of RotorS before. I think @burrimi would know a lot more about the state of the attitude controller.

burrimi, 2016-10-18:
Hi sytelus,
this is a leftover of our old controller factory implementation, which allowed to switch controllers easily but added quite a lot of complexity. We droped it because we wanted to make the code for rotors as simple as possible. In the process of removing the factory we forgot to delete these files. But they are never compiled or included in the cmake project and should not lead to any problems.

closed 2018-12-06 by ffurrer

AbhinayNS, 2020-05-11:
hi, 
i was actually trying to model the attitude control loop as a second order system and for that i wanted to give roll and pitch commands for the x and y axis control and position input for the z axis control. 
From what i have observed, none of the controller nodes are capable of doing this, so i wanted to create a attitude control node but again I faced the same issue of the those two header files not being available. Could anyone suggest me something on how do I go about this and is there already a way where in we can give roll and pitch commands for x and y axes and position or velocity command for the z axis.