ethz-asl/rotors_simulator #311: Correct way to reset a simulation for many consecutive runs

alexklibisz, 2016-10-18:
For context, I am trying to use a genetic algorithm to train a simple autopilot. This involves running many consecutive flights to achieve a correct configuration.

My question: What is the correct way to reset a quadcopter between multiple runs? Ideally this would reset the quadcopter to such a state that behavior is exactly the same for every run.

Some more details:

I've tried to implement this by calling the `/gazebo/reset_world` service from a C++ node:

```
std_srvs::Empty resetWorldSrv;
ros::service::call("/gazebo/reset_world", resetWorldSrv);
```

Calling the service resets the quadcopter back to position (0,0,0) and appears to have the desired behavior at first. But eventually the quadcopter rotors disappear and the quadcopter can no longer take off:

[image]

I think that this effect might have to do with the `max_contacts` param. I've tried to increase this setting to a higher value via `rosparam set /gazebo/max_contacts`, via `rosservice call /gazebo/set_physics_properties`, and on the C++ node handle via `nodeHandle.setParam()`. None of these prevent the disappearing rotors behavior.

Any tips towards a complete reset implementation are very appreciated!

alexklibisz, 2016-10-30:
I've since noticed that after becoming detached, the rotors end up directly at the origin underneath the MAV. You can see the blue and read underneath the MAV. I've noticed this problem occurs more quickly when applying faster rotor speeds (> 550).

shening, 2017-11-07:
Hey, did you ever figure this out? I am also trying to figure out a way to do multiple runs.

alexklibisz, 2017-11-07:
@shening I never figured out how to prevent it from falling apart. I did end up getting multiple runs to work by encapsulating the ROS setup in a docker container and starting/killing the container repeatedly. This didn't help much because there was no reliable  way to detect that the drone had fallen apart. Definitely not an ideal setup and I don't have access to the code anymore.

nkhedekar, 2018-08-23:
For me, calling the service as `rosservice call /gazebo/reset_world` from the command line is able to restart the simulation from the beginning. Can you try keeping a pose (other than 0,0,0) published and then calling the service?