ethz-asl/rotors_simulator #314: how to control the rotors using joystick?

jazsonma, 2016-10-28:
I try to use the joy_interface launch file and the joystick. however,it doesn't work.

pvechersky, 2016-10-28:
@jazsonma 

Earlier this week I started using a joystick to fly a MAV in RotorS and it worked fine for me.

- Are you using the 'mav_with_joy.launch' file?
- Do you get any error messages in your console when you run the launch file? (like device not found or missing files)
- Keep in mind that our launch files start Gazebo in paused mode. Make sure you un-pause the simulation or change the 'paused' flag to false in order for the MAV to fly.

jazsonma, 2016-10-31:
@pvechersky 
yes,when i change the 'paused' flag to false in the 'mav_with_joy.launch' file ,it can  fly.Thanks!

jazsonma, 2016-10-31:
@pvechersky 
yes, when I change the 'paused' flag to false ,it can fly

ffurrer, 2016-11-02:
Closing, as this seems to be resolved.

closed 2016-11-02 by ffurrer