ethz-asl/rotors_simulator #317: Topic frequency measurement issue (discrepancy between methods)

danyloM, 2016-11-18:
There is a weird discrepancy between the results of different ways of measuring the frequency with which messages are published over a topic. For example, consider the topics:

- `/hummingbird/ground_truth/odometry` publishing the true odometry of the hummingbird quad being simulated in Gazebo
- `/hummingbird/ground_truth/imu` publishing the true IMU data of the same quad

I used three methods to measure the message frequency:

1. `rostopic hz <topic name>`
2. I wrote [my own node](https://github.com/danyloM/rotors_simulator/blob/feature/danyloM_control_tests/rotors_danylo/src/nodes/frequency_probe_node.cpp) which subscribes to the topic, counts how many messages it received in one second and publishes this number every second
3. Count how many messages arrived in one second directly in the feedback controller node (in my case, [lines 142-166](https://github.com/danyloM/rotors_simulator/blob/feature/danyloM_control_tests/rotors_danylo/src/nodes/lqr_controller_node.cpp#L142))

Note that the parameter `use_sim_time` is set to `true` which means that normally all the timing is relative to the time given by the `/clock` topic which is published by `/gazebo` (all of this I checked and I can guarantee that this is what's happening at least for measurement methods 2 and 3 above - I don't know what's going on in method 1).

For a sample run of my simulation, I get the following:

### Method 1

```
$ rostopic hz /hummingbird/ground_truth/odometry 
subscribed to [/hummingbird/ground_truth/odometry]
WARNING: may be using simulated time
average rate: 100.000
	min: 0.000s max: 0.020s std dev: 0.00376s window: 100
average rate: 100.000
	min: 0.000s max: 0.020s std dev: 0.00401s window: 200
average rate: 100.000
	min: 0.000s max: 0.040s std dev: 0.01028s window: 298
average rate: 100.000
	min: 0.000s max: 0.040s std dev: 0.01244s window: 398
average rate: 100.000
	min: 0.000s max: 0.050s std dev: 0.01355s window: 500
average rate: 100.000
	min: 0.000s max: 0.050s std dev: 0.01425s window: 600
average rate: 100.000
	min: 0.000s max: 0.050s std dev: 0.01473s window: 700
average rate: 100.000
	min: 0.000s max: 0.050s std dev: 0.01508s window: 800
average rate: 100.000
	min: 0.000s max: 0.050s std dev: 0.01534s window: 900
average rate: 100.000
	min: 0.000s max: 0.050s std dev: 0.01555s window: 1000
average rate: 100.000
	min: 0.000s max: 0.050s std dev: 0.01572s window: 1100
average rate: 100.000
	min: 0.000s max: 0.050s std dev: 0.01586s window: 1200
average rate: 100.000
	min: 0.000s max: 0.050s std dev: 0.01598s window: 1300
average rate: 100.000
	min: 0.000s max: 0.050s std dev: 0.01608s window: 1400
```

```
$ rostopic hz /hummingbird/ground_truth/imu
subscribed to [/hummingbird/ground_truth/imu]
WARNING: may be using simulated time
average rate: 100.000
	min: 0.000s max: 0.020s std dev: 0.00142s window: 100
average rate: 100.000
	min: 0.000s max: 0.020s std dev: 0.00301s window: 200
average rate: 100.000
	min: 0.000s max: 0.020s std dev: 0.00317s window: 299
average rate: 100.000
	min: 0.000s max: 0.020s std dev: 0.00293s window: 397
average rate: 100.000
	min: 0.000s max: 0.020s std dev: 0.00269s window: 497
average rate: 100.000
	min: 0.000s max: 0.020s std dev: 0.00278s window: 597
average rate: 100.000
	min: 0.000s max: 0.020s std dev: 0.00284s window: 697
average rate: 100.000
	min: 0.000s max: 0.020s std dev: 0.00279s window: 798
average rate: 100.000
	min: 0.000s max: 0.020s std dev: 0.00275s window: 898
average rate: 100.000
	min: 0.000s max: 0.020s std dev: 0.00280s window: 998
average rate: 100.000
	min: 0.000s max: 0.020s std dev: 0.00273s window: 1098
```

If I let the commands output the average rate for a long time, it sometimes flickers up or down but by no more than 1 Hz.

**Conclusion from method 1**:  the `rostopic hz` method gives the confident impression that the topics `/hummingbird/ground_truth/odometry` and `/hummingbird/ground_truth/imu` output messages at 100 Hz.

### Method 2

[image]

**Conclusion from method 2**: subscribing to the `/hummingbird/ground_truth/odometry` and `/hummingbird/ground_truth/imu` methods, it appears that `imu` messages are received at more or less 100 Hz *but* `odometry` *messages are received at a quite variable rate of 25 to 65 Hz*.

### Method 3

Here I just have a callback function which outputs the `ros::Time::now().toSec()` at which it was called and the frequency (== number of messages) received in the last second. The console output is:

```
call time: t = 16.000
odometry:  freq = 37.000
imu:       freq = 100.000
call time: t = 17.000
odometry:  freq = 42.000
imu:       freq = 100.000
call time: t = 18.000
odometry:  freq = 39.000
imu:       freq = 100.000
call time: t = 19.000
odometry:  freq = 38.000
imu:       freq = 100.000
call time: t = 20.000
odometry:  freq = 43.000
imu:       freq = 99.000
call time: t = 21.000
odometry:  freq = 39.000
imu:       freq = 101.000
call time: t = 22.000
odometry:  freq = 33.000
imu:       freq = 100.000
call time: t = 23.000
odometry:  freq = 33.000
imu:       freq = 100.000
call time: t = 24.000
odometry:  freq = 40.000
imu:       freq = 100.000
call time: t = 25.000
odometry:  freq = 39.000
imu:       freq = 100.000
call time: t = 26.000
odometry:  freq = 38.000
imu:       freq = 100.000
call time: t = 27.000
odometry:  freq = 36.000
imu:       freq = 100.000
call time: t = 28.000
odometry:  freq = 41.000
imu:       freq = 100.000
call time: t = 29.000
odometry:  freq = 38.000
imu:       freq = 100.000
call time: t = 30.000
odometry:  freq = 41.000
imu:       freq = 100.000
```

**Conclusion from method 3**: the same as for Method 2.

### Steps to reproduce the results

Clone my branch of Rotors (found [here](https://github.com/danyloM/rotors_simulator/tree/feature/danyloM_control_tests)). Then you can in principle `catkin build` and do `roslaunch rotors_danylo lqr_controller.launch`. You will have method 3 printed to the screen immediately and you can use `rqt_plot` to plot the `/freq_probe/*` topics in method 2. Method 1 you can type in manually in the terminal.

### The Issue

Clearly, methods 2 and 3 (the ones that I implemented myself) don't agree with method 1 for the `/hummingbird/ground_truth/odometry` topic. I cannot understand why this is? Are my methods missing something, or is there a hidden quirk is `rostopic hz`?

At the end of the day, I am doing a controller design in RotorS. Since the implementation in the ROS nodes is in discrete time, I need to have an idea about the frequencies at which the sensors are sending messages and at which my controller is itself running. Please answer the following issues that the above analysis brings up:

1. Why does method 1 disagree with methods 2 and 3? Which conclusion is correct?
2. Why does the `imu` topic run at 100 Hz and the `odometry` topic at a wild 26 to 65 Hz (assuming methods 2 and 3 are the correct ones)?
3. Where are the frequencies at which the sensors output their measurements defined? I.e. for the odometry and the imu sensors.
4. At what frequency does the quadcopter simulation itself run? In other words, what is the maximum frequency at which I could imagine running my controller before I begin running into issues with the simulated quad itself not being simulated at a fast enough rate (I guess by Nyquist's criterion, my controller frequency should ideally be at least 5 times less (2 times theoretically) than the simulation frequency)?

dmalyuta, 2016-11-29:
Bump. Does really no one have an idea about this/has had this issue?

ZacharyTaylor, 2016-11-29:
Your code also has a queue size of 1, are you sure you are not dropping messages?

dmalyuta, 2016-11-29:
Can you please be more specific about which part of the code you are referring to? [Line 23 of the frequency probe](https://github.com/dmalyuta/rotors_simulator/blob/feature/danyloM_control_tests/rotors_danylo/src/nodes/frequency_probe_node.cpp#L23) or [Line 25 of the actual controller](https://github.com/dmalyuta/rotors_simulator/blob/feature/danyloM_control_tests/rotors_danylo/src/nodes/lqr_controller_node.cpp#L25)? In any case, how would I best check if messages are being dropped?

ZacharyTaylor, 2016-11-29:
If this is the issue then you would probably need to change it in both places. Not sure if you can detect dropped messages, probably easiest to set a big queue or run gazebo slower and see if the issue improves.

ttdm, 2017-04-10:
Hello, 

I have a strange behavior that might be linked to your tests : 
After a few secs, the Odometry message does not appear to be published any 0.01s anymore but it is published as 4 simultanious messages every 0.04s.
Below is a rosbag of a record of my odometry topic : 
[image]
the ardrone/odometry_ground_truth/odometry is the topic we are talking ( while the filtered one is me trying to use the ekf in robot_localization but not having any success ... )

You may indeed drop all the simultaneous messages in your handmade kernel, can you check that is in fact what's happening ? 
( Note that it would be coherent with your rostopic hz report stating : 
average rate: 100.000
	min: **0.000s** max: **0.040s** std dev: 0.01244s window: 398 
which i also have )

Anyone as an idee why this sort of behaviour could happen ?

ffurrer, 2017-05-03:
Closing this due to inactivity, please reopen if the problem is still there.

closed 2017-05-03 by ffurrer

danielhentzen, 2018-07-02:
I came across a similar issue while timing the delay between successive callbacks of the odometry subscriber in the default Lee position controller. There is a variable frequency between 0.01 and 0.05. After applying a tcpNoDelay() to the odometry subscriber, the issue goes away - the time elapsed between successive callbacks corresponds to the odometry rate as measured by hz.