ethz-asl/rotors_simulator #325: Investigate replacing existing custom GPS plugin with one provided by hector_gazebo_plugins

gbmhunter, 2016-12-12:
See [http://wiki.ros.org/hector_gazebo_plugins](http://wiki.ros.org/hector_gazebo_plugins).

ffurrer, 2016-12-12:
I think what @LorenzMeier meant (@LorenzMeier correct me if I'm wrong) was to look at the plugin shipped with newer Gazebo versions: https://bitbucket.org/osrf/gazebo/src/d8e129cd0e93e9e8e666269c5844b0b3b33eb975/gazebo/sensors/GpsSensor.cc?at=default&fileviewer=file-view-default
But yes, it's definitely worth comparing the three methods.

LorenzMeier, 2016-12-12:
Yes, I meant the Gazebo one. Our target version is Gazebo 7

gbmhunter, 2016-12-12:
Thanks for clearing that up.

jgoppert, 2017-01-20:
I think this should be closed in favor of https://github.com/ethz-asl/rotors_simulator/issues/340.

http://wiki.ros.org/hector_gazebo_plugins is closer to what we need but still doesn't have delay.

LorenzMeier, 2017-01-20:
We should use the official GPS plugins from Gazebo or build our own specialised version of a GPS sensor.