ethz-asl/rotors_simulator #330: "Undefined symbol" linking error when adding GazeboRosInterfacePlugin calls to the GazeboMultirotorBase class.

gbmhunter, 2016-12-15:
Error:

````
[Err] [Plugin.hh:165] Failed to load plugin librotors_gazebo_multirotor_base_plugin.so:
/home/gbmhunter/catkin_ws/devel/lib/librotors_gazebo_multirotor_base_plugin.so:
undefined symbol: _ZN6gazebo24GazeboRosInterfacePlugin8instanceE
````

gbmhunter, 2017-01-19:
Fixed.

closed 2017-01-19 by gbmhunter

Matnay, 2020-06-29:
Could you please let me know how you fixed this. I am having a similar issue and have not been able to find a workaround yet.
```
gzserver: symbol lookup error: /home/matnay/rotors_ws/devel/lib/librotors_gazebo_multirotor_base_plugin.so: undefined <br>symbol: _ZN14gz_sensor_msgs9ActuatorsC1Ev
```

guiaugustoga, 2020-12-22:
Ha

> Could you please let me know how you fixed this. I am having a similar issue and have not been able to find a workaround yet.
> 
> ```
> gzserver: symbol lookup error: /home/matnay/rotors_ws/devel/lib/librotors_gazebo_multirotor_base_plugin.so: undefined symbol: _ZN14gz_sensor_msgs9ActuatorsC1Ev
> ```
> 
> Thank you

Having the same issue. Any news ?