ethz-asl/rotors_simulator #344: Add namespace support for ROS interface plugin.

gbmhunter, 2017-02-02:
Currently the namespace is statically set to `firefly` in `GazeboRosInterfacePlugin.cpp`. However, the other plugins should specify the namespace of the topic they wish to connect by passing it into the `ConnectGazeboToRosMsg` or `ConnectRosToGazeboMsg`.

This was meant to be fixed in commit `30ec6fea1e404badf5df6c5c84f31dbb63f12155`, but for some reason these changes stopped the PX4 build/run command `make posix_sitl_default gazebo` from working correctly (Gazebo would hang when loading).

gbmhunter, 2017-02-10:
@ffurrer @burrimi @LorenzMeier 

Hey guys, this is one last thing I really think needs fixing before the pull requests can be merged. I have given the above issue another fix attempt and it now seems to be working, but I still do have one question...

Where Gazebo namespace should the two ROS interface plugin "config" topics belong under? 

Currently the two topics are at:
`/gazebo/firefly/connect_gazebo_to_ros_topic`
`/gazebo/firefly/connect_ros_to_gazebo_topic`

The ROS interface plugin is a world plugin, i.e. there only needs to be one ROS interface plugin per world (it can be used by multiple MAV's, other world plugins e.t.c). Should these two config topics be in `/gazebo/default/` instead?

ffurrer, 2017-02-11:
@gbmhunter I think if there is only one (world) plugin they should probably be somewhere in `/gazebo/rotors/...`, or `/gazebo/rotors_ros_config/`, or `/gazebo/rotors_ros_topic_registration`.
@burrimi, @LorenzMeier wdyt?

burrimi, 2017-02-13:
I would vote for the first '/gazebo/rotors/...'

gbmhunter, 2017-02-13:
O.K. I'll use `/gazebo/rotors/...`

gbmhunter, 2017-02-13:
Implemented in 960064382eeadb46bb29b7e2c41ba71bad65d61d.

closed 2017-02-13 by gbmhunter