ethz-asl/rotors_simulator #352: cmake problem - protobuf (with possible fix?)

ckchow, 2017-02-15:
I get a cmake problem when building on Ubuntu 16.04 / Kinetic:

```
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:298 (message):
  catkin_package() absolute include dir
  '/home/chris/projects/catkin_ws/devel/.private/rotors_gazebo_plugins/include/'
  does not exist
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:181 (catkin_package)
```

I've fixed it by moving the protobuf copy block in rotors_gazebo_plugins/CMakeLists.txt from the bottom to before catkin_package, but I'm not sure if the build problem is just because I'm on an unsupported platform. I would open a PR otherwise.

Michael9792, 2017-02-22:
My problem is the same as yours except the error line in CMakeLIsts.txt is not 181. My Ubuntu version is 14.04 and my ROS version is indigo. 
Seems that it is the problem of catkin. I tried to use the method [here](https://github.com/ethz-asl/glog_catkin/issues/15). After I changed my python-catkin-tools version to 0.3.1, there was another problem which said "PROTOBUF_PROTOC_EXECUTABLE-NOTFOUND: Command not found" just like #354 . I'm really confused now.

burrimi, 2017-02-22:
Do you have the protobuf compiler installed?
`sudo apt-get install protobuf-compiler`

marco-tranzatto, 2017-02-22:
Same problem here, with protobuf compiler installed in Ubuntu 14.04.
`Errors     << rotors_gazebo_plugins:make /home/marco/catkin_ws/logs/rotors_gazebo_plugins/build.make.005.log           
make[2]: PROTOBUF_PROTOC_EXECUTABLE-NOTFOUND: Command not found
make[2]: *** [Twist.pb.cc] Error 127
make[1]: *** [CMakeFiles/mav_msgs.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
`

hitimo, 2017-02-22:
Did you clean your workspace before rebuilding? (i.e. `rm -f ~/catkin_ws/build/ ~/catkin_ws/devel/`) I just tested this and `sudo apt-get install protobuf-compiler` solved it for me.

marco-tranzatto, 2017-02-24:
ok, this solved the problem for me! :smile:

Michael9792, 2017-02-24:
Great thanks! A complete rebuild solved the problem.

ffurrer, 2017-02-27:
@ckchow did a complete rebuild and the protobuf-compiler installation also work for you?

ckchow, 2017-02-27:
I actually had protobuf-compiler already installed, still get the problem mentioned even after doing `catkin clean` and `catkin build`. I'm pretty sure it's because the `catkin_package` macro at line 181 tries to include the compiled protobuf stuff at `${CATKIN_DEVEL_PREFIX}/include/${PACKAGE_NAME}` before it's created by the protobuf copy command at 441.

Starting @ 441
```
# Copy over all converted protobuf files (*.pb.h and *.pb.c) from the src directory to
# the devel directory. This is required so that you can include the protobuf header files from other C++ files.
if (NOT NO_ROS)
  file(COPY ${CMAKE_CURRENT_BINARY_DIR}/ DESTINATION ${CATKIN_DEVEL_PREFIX}/include/${PACKAGE_NAME} FILES_MATCHING PATTERN "*.pb.*")
endif()
```
Moving that block before line 181 solves the problem for me. I think if you catkin build twice the problem is masked because the copy from the previous run already touched the directory so that `catkin_package` can find it correctly. It's not a huge problem, but I wanted to report it if someone else ran into it.

gbmhunter, 2017-02-27:
@ckchow yes I have also experienced similar problems regarding protobuf files, with similar masking behavior if catkin is run twice (or other build processes).

@helenol wrote the initial `file(...)` magic to get the .protobuf files in the right place, perhaps she can help here?

zgqball, 2017-03-09:
many thanks !  i solved the same problem with your help!!!!