ethz-asl/rotors_simulator #354: catkin build error

ptrkfry, 2017-02-22:
Hi,
I compiled rotors from scratch and get the following error:
Any ideas?

```
patrik@patrik-Inspiron-1564:~/catkin_ws$ catkin build rotors_gazebo
-----------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/indigo
Workspace:                   /home/patrik/catkin_ws
-----------------------------------------------------------
Source Space:       [exists] /home/patrik/catkin_ws/src
Log Space:         [missing] /home/patrik/catkin_ws/logs
Build Space:        [exists] /home/patrik/catkin_ws/build
Devel Space:        [exists] /home/patrik/catkin_ws/devel
Install Space:      [unused] /home/patrik/catkin_ws/install
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-----------------------------------------------------------
[build] Found '45' packages in 0.0 seconds.                                                                                                      
[build] Updating package table.                                                                                                                  
Starting  >>> mav_msgs                                                                                                                           
Starting  >>> rotors_comm                                                                                                                        
Finished  <<< rotors_comm                          [ 4.3 seconds ]                                                                               
Finished  <<< mav_msgs                             [ 5.3 seconds ]                                                                               
Starting  >>> rotors_control                                                                                                                     
Finished  <<< rotors_control                       [ 25.0 seconds ]                                                                              
Starting  >>> rotors_gazebo_plugins                                                                                                              
_________________________________________________________________________________________________________________________________________________
Errors     << rotors_gazebo_plugins:make /home/patrik/catkin_ws/logs/rotors_gazebo_plugins/build.make.000.log                                    
make[2]: PROTOBUF_PROTOC_EXECUTABLE-NOTFOUND: Command not found
make[2]: *** [Wrench.pb.cc] Error 127
make[1]: *** [CMakeFiles/mav_msgs.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/patrik/catkin_ws/build/rotors_gazebo_plugins; catkin build --get-env rotors_gazebo_plugins | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.................................................................................................................................................
Failed     << rotors_gazebo_plugins:make           [ Exited with code 2 ]                                                                        
Failed    <<< rotors_gazebo_plugins                [ 4.2 seconds ]                                                                               
Abandoned <<< rotors_gazebo                        [ Unrelated job failed ]                                                                      
[build] Summary: 3 of 5 packages succeeded.                                                                                                      
[build]   Ignored:   40 packages were skipped or are blacklisted.                                                                                
[build]   Warnings:  None.                                                                                                                       
[build]   Abandoned: 1 packages were abandoned.                                                                                                  
[build]   Failed:    1 packages failed.                                                                                                          
[build] Runtime: 34.7 seconds total.                                                                                                             
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

hitimo, 2017-02-22:
```
sudo apt-get install protobuf-compiler
cd catkin_ws
rm -rf build/ devel/
catkin build 
```
should solve it.

closed 2017-02-23 by hitimo

SaiManikanta23, 2017-04-14:
Hello I am trying to follows the steps provided and got the following build problem, can you help me with it please?? and is this because I am using ros-kinetic?

bash: cd: catkin_ws: No such file or directory
sai@sai-VirtualBox:~/catkin_ws$ rm -rf build/ devel/
sai@sai-VirtualBox:~/catkin_ws$ catkin build
--------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /home/sai/catkin_ws
--------------------------------------------------------
Source Space:       [exists] /home/sai/catkin_ws/src
Log Space:          [exists] /home/sai/catkin_ws/logs
Build Space:        [exists] /home/sai/catkin_ws/build
Devel Space:        [exists] /home/sai/catkin_ws/devel
Install Space:      [unused] /home/sai/catkin_ws/install
DESTDIR:            [unused] None
--------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------
[build] Found '12' packages in 0.0 seconds.                                    
[build] Updating package table.                                                
Warning: generated devel space setup files have been deleted.
Starting  >>> catkin_tools_prebuild                                            
_______________________________________________________________________________
Errors     << catkin_tools_prebuild:cmake /home/sai/catkin_ws/logs/catkin_tools_prebuild/build.cmake.003.log
CMake Error at /opt/ros/kinetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/local/bin/python
  "/opt/ros/kinetic/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/kinetic/share/catkin/cmake/../package.xml"
  "/home/sai/catkin_ws/build/catkin_tools_prebuild/catkin/catkin_generated/version/package.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package_xml.cmake:63 (safe_execute_process)
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:151 (_catkin_package_xml)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

cd /home/sai/catkin_ws/build/catkin_tools_prebuild; catkin build --get-env catkin_tools_prebuild | catkin env -si  /usr/bin/cmake /home/sai/catkin_ws/build/catkin_tools_prebuild --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/sai/catkin_ws/devel/.private/catkin_tools_prebuild -DCMAKE_INSTALL_PREFIX=/home/sai/catkin_ws/install; cd -
...............................................................................
Failed     << catkin_tools_prebuild:cmake          [ Exited with code 1 ]      
Failed    <<< catkin_tools_prebuild                [ 2.6 seconds ]             
Abandoned <<< mav_msgs                             [ Unrelated job failed ]    
Abandoned <<< planning_msgs                        [ Unrelated job failed ]    
Abandoned <<< rotors_comm                          [ Unrelated job failed ]    
Abandoned <<< rotors_control                       [ Unrelated job failed ]    
Abandoned <<< rotors_description                   [ Unrelated job failed ]    
Abandoned <<< rotors_evaluation                    [ Unrelated job failed ]    
Abandoned <<< rotors_gazebo_plugins                [ Unrelated job failed ]    
Abandoned <<< rotors_gazebo                        [ Unrelated job failed ]    
Abandoned <<< rotors_hil_interface                 [ Unrelated job failed ]    
Abandoned <<< rotors_joy_interface                 [ Unrelated job failed ]    
[build] Summary: 0 of 11 packages succeeded.                                   
[build]   Ignored:   2 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 10 packages were abandoned.                               
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 2.7 seconds total.