ethz-asl/rotors_simulator #355: catkin build error: catkin_package() absolute include dir

Michael9792, 2017-02-24:
The process of `catkin build` won't succeed. The error says:
```
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:291 (message):
  catkin_package() absolute include dir
  '/home/michaelxu/catkin_ethz_asl/devel/rotors_gazebo_plugins/include/' does
  not exist
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:98 (_catkin_package)
  CMakeLists.txt:181 (catkin_package)
-- Configuring incomplete, errors occurred!

Failed to build package 'rotors_gazebo_plugins' because the following command:

# Command to reproduce:
cd /home/michaelxu/catkin_ethz_asl/build/rotors_gazebo_plugins && /home/michaelxu/catkin_ethz_asl/build/rotors_gazebo_plugins/build_env.sh /usr/bin/cmake /home/michaelxu/catkin_ethz_asl/src/rotors_simulator/rotors_gazebo_plugins -DCATKIN_DEVEL_PREFIX=/home/michaelxu/catkin_ethz_asl/devel/rotors_gazebo_plugins -DCMAKE_INSTALL_PREFIX=/home/michaelxu/catkin_ethz_asl/install; cd -

# Path to log:
cat /home/michaelxu/catkin_ethz_asl/build/build_logs/rotors_gazebo_plugins.log

Exited with return code: 1 
```

I use Ubuntu 14.04 and ROS Indigo. My `ros-indigo-catkin` version is 0.6.18. My `protobuf-compiler` version is 2.5.0. 

However, after I deleted `${CATKIN_DEVEL_PREFIX}/include/${PACKAGE_NAME}` [here](https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo_plugins/CMakeLists.txt#L182), the process of `catkin build` succeeded. 

So is there any problem with this line? Thanks for all the developers!

ffurrer, 2017-02-27:
@Michael9792 Is this resolved?

Michael9792, 2017-02-28:
@ffurrer Yes, `catkin build` succeeded after I had deleted ${CATKIN_DEVEL_PREFIX}/include/${PACKAGE_NAME} [here](https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo_plugins/CMakeLists.txt#L182). 

So my point is that whether we should delete ${CATKIN_DEVEL_PREFIX}/include/${PACKAGE_NAME}. In other word, is this a bug?

hitimo, 2017-03-07:
Have you tried 
```
cd  /home/michaelxu/catkin_ethz_asl/
catkin init
catkin config --merge-devel # Necessary for catkin_tools >= 0.4.
catkin config --extend /opt/ros/indigo
catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release
```
_before_ compiling?

Michael9792, 2017-03-07:
Wow! It works! Great thanks @hitimo! My python_catkin_tools version is 0.4.4, so I need to add `catkin config --merge-devel`. 
Could you please tell me why this line is necessary for catkin_tools version>=0.4? I only know the meaning of `catkin config --merge-devel` from [here](http://catkin-tools.readthedocs.io/en/latest/verbs/catkin_config.html).
Anything will be very helpful!

jgoppert, 2017-03-20:
I can confirm this fixes the build issue, but since merge-devel isn't default, can we fix this so that it is less cumbersome to build?

pulver22, 2017-03-20:
I have the same issue but when following @hitimo suggestions I got the following message:

> Error: The current devel space layout, `linked`,is incompatible with the configured layout, `merged`.

amrii1, 2017-04-06:
i use ubuntu 16.04 and ros kinetic, i had the same problem, but now i'm getting this error when trying to build the gazebo_plugins
[image]
thanks :)

shelderzhang, 2017-07-11:
```
sudo apt-get update
sudo apt-get install libgoogle-glog-dev
catkin clean
catkin build
```