ethz-asl/rotors_simulator #356: Cannot build RotorS package

tonyvanhain, 2017-03-05:
I want to try using RotorS and I have followed the steps to install it. When I try to run catkin build after doing the steps before, I got a lot of error messages. I am currently using ROS Kinetic. Can anyone help with this problem? 

`Errors     << rotors_hil_interface:make /home/anthonyivan/rotors_ws/logs/rotors_hil_interface/build.make.000.log
In file included from /home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_state_level_interface.cpp:17:0:
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:105:3: error: ‘mavlink_hil_gps_t’ does not name a type
   mavlink_hil_gps_t hil_gps_msg_;
   ^
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:108:3: error: ‘mavlink_hil_sensor_t’ does not name a type
   mavlink_hil_sensor_t hil_sensor_msg_;
   ^
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:130:3: error: ‘mavlink_hil_state_quaternion_t’ does not name a type
   mavlink_hil_state_quaternion_t hil_state_qtrn_msg_;
   ^
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_state_level_interface.cpp: In member function ‘virtual std::vector<mavros_msgs::Mavlink_<std::allocator<void> > > rotors_hil::HilStateLevelInterface::CollectData()’:
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_state_level_interface.cpp:70:3: error: ‘mavlink_message_t’ was not declared in this scope
   mavlink_message_t mmsg;
   ^
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_state_level_interface.cpp:70:3: note: suggested alternatives:
In file included from /opt/ros/kinetic/include/mavlink/v2.0/common/../message.hpp:15:0,
                 from /opt/ros/kinetic/include/mavlink/v2.0/common/common.hpp:16,
                 from /home/anthonyivan/catkin_ws/src/mavros/libmavconn/include/mavconn/mavlink_dialect.h:23,
                 from /home/anthonyivan/catkin_ws/src/mavros/mavros_msgs/include/mavros_msgs/mavlink_convert.h:18,
                 from /home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:23,
                 from /home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_state_level_interface.cpp:17:
/opt/ros/kinetic/include/mavlink/v2.0/common/../mavlink_types.h:121:4: note:   ‘mavlink::mavlink_message_t’
 }) mavlink_message_t;
    ^
/opt/ros/kinetic/include/mavlink/v2.0/common/../mavlink_types.h:121:4: note:   ‘mavlink::mavlink_message_t’
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_state_level_interface.cpp:82:3: error: ‘hil_state_qtrn_msg_’ was not declared in this scope
   hil_state_qtrn_msg_.time_usec = time_usec;
   ^
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_state_level_interface.cpp:102:3: error: ‘mavlink_hil_state_quaternion_t’ was not declared in this scope
   mavlink_hil_state_quaternion_t* hil_state_qtrn_msg_ptr = &hil_state_qtrn_msg_;
   ^
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_state_level_interface.cpp:102:35: error: ‘hil_state_qtrn_msg_ptr’ was not declared in this scope
   mavlink_hil_state_quaternion_t* hil_state_qtrn_msg_ptr = &hil_state_qtrn_msg_;
                                   ^
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_state_level_interface.cpp:103:50: error: ‘mmsg’ was not declared in this scope
   mavlink_msg_hil_state_quaternion_encode(1, 0, &mmsg, hil_state_qtrn_msg_ptr);
                                                  ^
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_state_level_interface.cpp:103:78: error: ‘mavlink_msg_hil_state_quaternion_encode’ was not declared in this scope
   mavlink_msg_hil_state_quaternion_encode(1, 0, &mmsg, hil_state_qtrn_msg_ptr);
                                                                              ^
In file included from /home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:17:0:
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:105:3: error: ‘mavlink_hil_gps_t’ does not name a type
   mavlink_hil_gps_t hil_gps_msg_;
   ^
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:108:3: error: ‘mavlink_hil_sensor_t’ does not name a type
   mavlink_hil_sensor_t hil_sensor_msg_;
   ^
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:130:3: error: ‘mavlink_hil_state_quaternion_t’ does not name a type
   mavlink_hil_state_quaternion_t hil_state_qtrn_msg_;
   ^
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp: In member function ‘virtual std::vector<mavros_msgs::Mavlink_<std::allocator<void> > > rotors_hil::HilSensorLevelInterface::CollectData()’:
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:89:3: error: ‘mavlink_message_t’ was not declared in this scope
   mavlink_message_t mmsg;
   ^
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:89:3: note: suggested alternatives:
In file included from /opt/ros/kinetic/include/mavlink/v2.0/common/../message.hpp:15:0,
                 from /opt/ros/kinetic/include/mavlink/v2.0/common/common.hpp:16,
                 from /home/anthonyivan/catkin_ws/src/mavros/libmavconn/include/mavconn/mavlink_dialect.h:23,
                 from /home/anthonyivan/catkin_ws/src/mavros/mavros_msgs/include/mavros_msgs/mavlink_convert.h:18,
                 from /home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:23,
                 from /home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:17:
/opt/ros/kinetic/include/mavlink/v2.0/common/../mavlink_types.h:121:4: note:   ‘mavlink::mavlink_message_t’
 }) mavlink_message_t;
    ^
/opt/ros/kinetic/include/mavlink/v2.0/common/../mavlink_types.h:121:4: note:   ‘mavlink::mavlink_message_t’
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:105:5: error: ‘hil_gps_msg_’ was not declared in this scope
     hil_gps_msg_.time_usec = time_usec;
[... 23 lines cut ...]
   mavlink_msg_hil_sensor_encode(1, 0, &mmsg, hil_sensor_msg_ptr);
                                        ^
/home/anthonyivan/rotors_ws/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:148:64: error: ‘mavlink_msg_hil_sensor_encode’ was not declared in this scope
   mavlink_msg_hil_sensor_encode(1, 0, &mmsg, hil_sensor_msg_ptr);
                                                                ^
make[2]: *** [CMakeFiles/rotors_hil_interface.dir/src/hil_state_level_interface.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/rotors_hil_interface.dir/src/hil_sensor_level_interface.cpp.o] Error 1
make[1]: *** [CMakeFiles/rotors_hil_interface.dir/all] Error 2
make: *** [all] Error 2
cd /home/anthonyivan/rotors_ws/build/rotors_hil_interface; catkin build --get-env rotors_hil_interface | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..............................................................................
Failed     << rotors_hil_interface:make            [ Exited with code 2 ]     
Failed    <<< rotors_hil_interface                 [ 12.6 seconds ]           
Abandoned <<< rotors_gazebo_plugins                [ Unrelated job failed ]   
Abandoned <<< rotors_gazebo                        [ Unrelated job failed ]   
Finished  <<< rotors_control                       [ 23.0 seconds ]           
[build] Summary: 8 of 11 packages succeeded.                                  
[build]   Ignored:   2 packages were skipped or are blacklisted.              
[build]   Warnings:  4 packages succeeded with warnings.                      
[build]   Abandoned: 2 packages were abandoned.                               
[build]   Failed:    1 packages failed.                                       
[build] Runtime: 44.5 seconds total.                                          
[build] Note: Workspace packages have changed, please re-source setup files to use them.
`

hitimo, 2017-03-06:
Have you compiled mavlink/mavros in your catkin workspace beforehand? e.g
```
cd ~/catkin_ws/src
git clone -b indigo-devel git@github.com:mavlink/mavros.git
git clone -b debian/kinetic/mavlink git@github.com:mavlink/mavlink-gbp-release.git
catkin build mavros --cmake-args -DCMAKE_BUILD_TYPE=Release -DMAVLINK_DIALECT=ASLUAV
```

hitimo, 2017-03-06:
@burrimi @ffurrer The hil_interface plugin should probably be optional for those who don't want/need it?

ckchow, 2017-03-09:
@tonyvanhain I have had lots of problems building hil_interface on kinetic, for me it has been safe to just `cd rotors_hil_interface && touch CATKIN_IGNORE` and skip building the plugin. I think it's related to a weird C/C++ API conflict in mavros kinetic. I haven't tried building it with the directions above, but if you don't need the HIL plugin it should be safe to skip.

tonyvanhain, 2017-03-10:
@hitimo the solution that @ckchow gave works. Thank you for the help!

tonyvanhain, 2017-03-10:
@hitimo if I dont use the HIL plugin, then I will not be able to use the HIL simulation?

hitimo, 2017-03-10:
@tonyvanhain Yes, you need the HIL plugin for HIL simulation.

tonyvanhain, 2017-03-11:
@hitimo I was planning on using HIL simulation for my project. Ok then thank you for the information

closed 2017-03-11 by tonyvanhain

TSC21, 2017-04-02:
@vooon ping! Having the same problem here.

TSC21, 2017-04-02:
Adding '#include "common/mavlink.h" ' to 'hil_interface.h' solved the problem. Will send a PR.

TSC21, 2017-04-13:
The issue is related with the way MAVROS currently handles mavlink msgs (right now it uses a C++ wrapper and not C). So after https://github.com/mavlink/mavros/pull/684 is merged, I will update the `rotors_hil_interface` to work with the new HIL plugin of mavros and simplify the current implementation.

TSC21, 2017-08-17:
Update: mavlink/mavros#684 merged, which means that the interface can be updated also.

cehberlin, 2017-08-29:
@TSC21 Hi, I am experiencing the same problem. I just tried to build with ROS kinetic and a clone MAVROS master branch in my workspace, but still have the issue.

TSC21, 2017-08-29:
> @TSC21 Hi, I am experiencing the same problem. I just tried to build with ROS kinetic and a clone MAVROS master branch in my workspace, but still have the issue.

When I said that the interface can be updated, it means that we now have a way of interfacing with the FCU on HIL mode through MAVROS completly, given that now we support all the the HIL msgs used on the Mavlink protocol. "Can be updated" means that this needs to be updated to support what MAVROS can do at the moment, and allow `rotors_simulator` to be able to use the latest MAVROS+Mavlink version.

cehberlin, 2017-08-29:
I see! Thanks for the clarification!

TSC21, 2017-08-29:
I have not purposed a PR for updating `rotors_hil_interface` because I didn't had time for it yet.

juanmed, 2018-11-29:
Hello, I am late for this issue but I am facing the same issues. Trying to catking make RotorS fails not finding the correct files for the HIL Interface. Ignoring the whole rotors_hil_interface folder compiles but I need the HIL interface, and so far have been unable to make even after including indigo-devel version. 

Any updates on the correct procedure to install RotorS WITH HIL Interface?

juanmed, 2018-11-29:
Hello, this solution works:

https://github.com/ethz-asl/rotors_simulator/pull/522/files?fbclid=IwAR2SxViGule-u8S4A1kE0L6HCwW3cixtO7IgMBNTp9TILT62TGdXPlsxApw