ethz-asl/rotors_simulator #358: Problem with mav_hovering_example.launch example

ptrkfry, 2017-03-08:
I tried to launch the mav_hovering_example.launch file with the yosemite world. The world opens and the MAV is shown, but it doesn't hover. The propellers don't even turn. When I try it with the basic world it works (see output for both cases below). The problem seems to have something to do with the function "void gazebo::GazeboMotorModel::CreatePubsAndSubs()" as this doesn't seem to get the same output in the working example as in the failing one. 
Maybe some sort of a timing problem? My laptop is not the newest (only i3 processor).

Fail with yosemite world:

`patrik@patrik-Inspiron-1564:~/catkin_ws$ roslaunch rotors_gazebo mav_hovering_example.launch mav_name:=firefly world_name:=yosemite
... logging to /home/patrik/.ros/log/c7bc68a6-fde5-11e6-8a17-f07bcb56532d/roslaunch-patrik-Inspiron-1564-21878.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://patrik-Inspiron-1564:34818/

SUMMARY
========

PARAMETERS
 * /firefly/lee_position_controller_node/angular_rate_gain/x: 0.52
 * /firefly/lee_position_controller_node/angular_rate_gain/y: 0.52
 * /firefly/lee_position_controller_node/angular_rate_gain/z: 0.18
 * /firefly/lee_position_controller_node/attitude_gain/x: 3
 * /firefly/lee_position_controller_node/attitude_gain/y: 3
 * /firefly/lee_position_controller_node/attitude_gain/z: 0.15
 * /firefly/lee_position_controller_node/inertia/xx: 0.0347563
 * /firefly/lee_position_controller_node/inertia/xy: 0.0
 * /firefly/lee_position_controller_node/inertia/xz: 0.0
 * /firefly/lee_position_controller_node/inertia/yy: 0.0458929
 * /firefly/lee_position_controller_node/inertia/yz: 0.0
 * /firefly/lee_position_controller_node/inertia/zz: 0.0977
 * /firefly/lee_position_controller_node/mass: 1.56779
 * /firefly/lee_position_controller_node/position_gain/x: 6
 * /firefly/lee_position_controller_node/position_gain/y: 6
 * /firefly/lee_position_controller_node/position_gain/z: 6
 * /firefly/lee_position_controller_node/rotor_configuration/0/angle: 0.52359877559
 * /firefly/lee_position_controller_node/rotor_configuration/0/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/0/direction: 1.0
 * /firefly/lee_position_controller_node/rotor_configuration/0/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/0/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/1/angle: 1.57079632679
 * /firefly/lee_position_controller_node/rotor_configuration/1/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/1/direction: -1.0
 * /firefly/lee_position_controller_node/rotor_configuration/1/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/1/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/2/angle: 2.61799387799
 * /firefly/lee_position_controller_node/rotor_configuration/2/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/2/direction: 1.0
 * /firefly/lee_position_controller_node/rotor_configuration/2/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/2/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/3/angle: -2.61799387799
 * /firefly/lee_position_controller_node/rotor_configuration/3/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/3/direction: -1.0
 * /firefly/lee_position_controller_node/rotor_configuration/3/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/3/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/4/angle: -1.57079632679
 * /firefly/lee_position_controller_node/rotor_configuration/4/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/4/direction: 1.0
 * /firefly/lee_position_controller_node/rotor_configuration/4/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/4/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/5/angle: -0.52359877559
 * /firefly/lee_position_controller_node/rotor_configuration/5/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/5/direction: -1.0
 * /firefly/lee_position_controller_node/rotor_configuration/5/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/5/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/velocity_gain/x: 4.7
 * /firefly/lee_position_controller_node/velocity_gain/y: 4.7
 * /firefly/lee_position_controller_node/velocity_gain/z: 4.7
 * /firefly/robot_description: <?xml version="1....
 * /firefly/tf_prefix: 
 * /rosdistro: indigo
 * /rosversion: 1.11.20
[... 251 lines cut ...]
[ INFO] [1488999710.424433075, 0.030000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1488999710.507090532, 0.110000000]: Physics dynamic reconfigure ready.
Warning [Publisher.cc:134] Queue limit reached for topic /gazebo/default/firefly/odometry_sensor1/pose, deleting message. This warning is printed only once.
Warning [Publisher.cc:134] Queue limit reached for topic /gazebo/default/firefly/odometry_sensor1/pose_with_covariance, deleting message. This warning is printed only once.
Warning [Publisher.cc:134] Queue limit reached for topic /gazebo/default/firefly/odometry_sensor1/position, deleting message. This warning is printed only once.
Warning [Publisher.cc:134] Queue limit reached for topic /gazebo/default/firefly/odometry_sensor1/transform, deleting message. This warning is printed only once.
Warning [Publisher.cc:134] Queue limit reached for topic /gazebo/default/firefly/odometry_sensor1/odometry, deleting message. This warning is printed only once.
Warning [Publisher.cc:134] Queue limit reached for topic /gazebo/default/broadcast_transform, deleting message. This warning is printed only once.
Warning [Publisher.cc:134] Queue limit reached for topic /gazebo/default/firefly/command/motor_speed, deleting message. This warning is printed only once.
[ INFO] [1488999715.445669955, 5.010000000]: Publishing waypoint on namespace //firefly: [0.000000, 0.000000, 1.000000].
^C[firefly/joint_state_publisher-8] killing on exit
[firefly/robot_state_publisher-7] killing on exit
[firefly/hovering_example-6] killing on exit
[firefly/lee_position_controller_node-5] killing on exit
[gazebo-2] killing on exit
[gazebo_gui-3] killing on exit
pure virtual method called
terminate called without an active exception
Aborted (core dumped)
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

burrimi, 2017-03-09:
on slow machines it can happen that the waypoint is published before everything is ready. Can you try the following command after launching the hovering example and waiting for a while:
`rosrun rotors_gazebo waypoint_publisher 0 0 1 0 __ns:=firefly`

ptrkfry, 2017-03-09:
Thanks for your answer. Unfortunately it didn't change anything. The waypoint_publisher starts, but the MAV doesn't move.

`patrik@patrik-Inspiron-1564:~/catkin_ws$ rosrun rotors_gazebo waypoint_publisher 0 0 1 0 __ns:=firefly
[ INFO] [1489089661.401069453]: Started waypoint_publisher.
[ INFO] [1489089662.595024281, 40.700000000]: Publishing waypoint on namespace /firefly: [0.000000, 0.000000, 1.000000].`

I also tried the following, with no result.

`rostopic pub /firefly/command/motor_speed mav_msgs/Actuators '{angular_velocities: [100, 100, 100, 100, 100, 100]}'`

 Any other ideas?

pvechersky, 2017-03-10:
Hi @ptrkfry,

I tested 'mav_hovering_example.launch' with Yosemite world on my machine and it works ... sometimes ... I am getting inconsistent behavior where it sometimes hovers and sometimes not, sometimes starts in paused state and sometimes not. But it is definitely capable of working :)

Several things to look into/try:

1) Verify that you don't have to manually un-pause the simulation.

2) In my latest pull request (which is not merged yet), I actually modified the Yosemite world/model a little bit. I have previously noticed an occasional discrepancy between the visual and collision layers in the Heightmap objects (I would sometimes be flying through seeminly free space and suddenly stop/crash as if I hit an obstacle). Not 100% sure, but I think it works better with my new model description. I'm attaching the new files here so that you can go ahead and try with the new Yosemite world and model. Just replace the following files with new ones:
- (rotors_gazebo)/worlds/yosemite.world
- (rotors_gazebo)/models/yosemite/yosemite.sdf
- (rotors_gazebo)/models/yosemite/materials/textures/yosemite.png
[yosemite.tar.gz](https://github.com/ethz-asl/rotors_simulator/files/833401/yosemite.tar.gz)

3) I noticed from the console output that you're using quite an old version of Gazebo, the version which comes with ROS Indigo installation. While the fact that it works with Basic world would suggest that this is not an issue, the one big difference with the Yosemite world is that it uses a Heightmap, so I'm not sure if the behavior for it changed somehow between versions. In developing RotorS we usually work with Gazebo version 5 or newer, so if updating Gazebo version is an option maybe you can explore that possibility ...

ptrkfry, 2017-03-15:
I would love to do that and also try out the tips from pvechersky, but unfortunately I can't seem to get rotors_simulator to build correctly again (I did a make clean and installed some other packages in the meantime). Now it fails at rotors_gazebo_plugins:

`This file was generated by an older version of protoc which is error incompatible with your Protocol Buffer headers. Please regenerate this file with a newer version of protoc.`

How do I do this?

Full output:

`patrik@patrik-Inspiron-1564:~/catkin_ws$ catkin build rotors_simulator
-----------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/indigo
Workspace:                   /home/patrik/catkin_ws
-----------------------------------------------------------
Source Space:       [exists] /home/patrik/catkin_ws/src
Log Space:          [exists] /home/patrik/catkin_ws/logs
Build Space:        [exists] /home/patrik/catkin_ws/build
Devel Space:        [exists] /home/patrik/catkin_ws/devel
Install Space:      [unused] /home/patrik/catkin_ws/install
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------
[build] Found '178' packages in 0.0 seconds.                                                                                           
[build] Package table is up to date.                                                                                                   
Starting  >>> catkin_simple                                                                                                            
Starting  >>> mav_msgs                                                                                                                 
Starting  >>> rotors_comm                                                                                                              
Starting  >>> rotors_description                                                                                                       
Finished  <<< rotors_description                   [ 1.2 seconds ]                                                                     
Starting  >>> rotors_evaluation                                                                                                        
Finished  <<< rotors_comm                          [ 1.7 seconds ]                                                                     
Finished  <<< mav_msgs                             [ 1.7 seconds ]                                                                     
Starting  >>> rotors_control                                                                                                           
Starting  >>> rotors_hil_interface                                                                                                     
Finished  <<< catkin_simple                        [ 1.3 seconds ]                                                                     
Starting  >>> rotors_joy_interface                                                                                                     
Finished  <<< rotors_evaluation                    [ 1.2 seconds ]                                                                     
Starting  >>> opencv3_catkin                                                                                                           
Finished  <<< rotors_hil_interface                 [ 1.2 seconds ]                                                                     
Finished  <<< rotors_control                       [ 1.7 seconds ]                                                                     
Finished  <<< rotors_joy_interface                 [ 1.6 seconds ]                                                                     
Finished  <<< opencv3_catkin                       [ 1.4 seconds ]                                                                     
Starting  >>> cv_bridge                                                                                                                
Finished  <<< cv_bridge                            [ 1.3 seconds ]                                                                     
Starting  >>> rotors_gazebo_plugins                                                                                                    
_______________________________________________________________________________________________________________________________________
Errors     << rotors_gazebo_plugins:make /home/patrik/catkin_ws/logs/rotors_gazebo_plugins/build.make.002.log                          
In file included from /home/patrik/catkin_ws/build/rotors_gazebo_plugins/MagneticField.pb.h:27:0,
                 from /home/patrik/catkin_ws/build/rotors_gazebo_plugins/MagneticField.pb.cc:5:
[... cut ...]
[build]   Ignored:   166 packages were skipped or are blacklisted.                                                                     
[build]   Warnings:  None.                                                                                                             
[build]   Abandoned: 1 packages were abandoned.                                                                                        
[build]   Failed:    1 packages failed.                                                                                                
[build] Runtime: 17.9 seconds total.  `

pvechersky, 2017-03-16:
I recall having some similar issue and needing to clean my workspace and then rebuild the first time I pulled the latest code with all the Protobuf messages. So if you haven't done so already, I would try to clean the workspace (at least all the RotorS packages) and do a fresh build.

JD-ETH, 2017-05-15:
I am experiencing the same issue here. This seems to happen in any world except the basic, the rotor doesn't spin for any of the quadcopter models. I hope this can be resolved soon

JD-ETH, 2017-05-15:
This is the result for running ` roslaunch rotors_gazebo mav_hovering_example.launch mav_name:=firefly world_name:=empty
`
under verbose. Has someone had similar issue?

`... logging to /home/jd/.ros/log/38c6bb74-39a4-11e7-b4b3-e4a7a00064b9/roslaunch-jd-ThinkPad-T460p-16840.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jd-ThinkPad-T460p:34038/

SUMMARY
========

PARAMETERS
 * /firefly/lee_position_controller_node/angular_rate_gain/x: 0.52
 * /firefly/lee_position_controller_node/angular_rate_gain/y: 0.52
 * /firefly/lee_position_controller_node/angular_rate_gain/z: 0.18
 * /firefly/lee_position_controller_node/attitude_gain/x: 3
 * /firefly/lee_position_controller_node/attitude_gain/y: 3
 * /firefly/lee_position_controller_node/attitude_gain/z: 0.15
 * /firefly/lee_position_controller_node/inertia/xx: 0.0347563
 * /firefly/lee_position_controller_node/inertia/xy: 0.0
 * /firefly/lee_position_controller_node/inertia/xz: 0.0
 * /firefly/lee_position_controller_node/inertia/yy: 0.0458929
 * /firefly/lee_position_controller_node/inertia/yz: 0.0
 * /firefly/lee_position_controller_node/inertia/zz: 0.0977
 * /firefly/lee_position_controller_node/mass: 1.56779
 * /firefly/lee_position_controller_node/position_gain/x: 6
 * /firefly/lee_position_controller_node/position_gain/y: 6
 * /firefly/lee_position_controller_node/position_gain/z: 6
 * /firefly/lee_position_controller_node/rotor_configuration/0/angle: 0.52359877559
 * /firefly/lee_position_controller_node/rotor_configuration/0/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/0/direction: 1.0
 * /firefly/lee_position_controller_node/rotor_configuration/0/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/0/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/1/angle: 1.57079632679
 * /firefly/lee_position_controller_node/rotor_configuration/1/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/1/direction: -1.0
 * /firefly/lee_position_controller_node/rotor_configuration/1/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/1/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/2/angle: 2.61799387799
 * /firefly/lee_position_controller_node/rotor_configuration/2/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/2/direction: 1.0
 * /firefly/lee_position_controller_node/rotor_configuration/2/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/2/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/3/angle: -2.61799387799
 * /firefly/lee_position_controller_node/rotor_configuration/3/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/3/direction: -1.0
 * /firefly/lee_position_controller_node/rotor_configuration/3/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/3/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/4/angle: -1.57079632679
 * /firefly/lee_position_controller_node/rotor_configuration/4/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/4/direction: 1.0
 * /firefly/lee_position_controller_node/rotor_configuration/4/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/4/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/5/angle: -0.52359877559
 * /firefly/lee_position_controller_node/rotor_configuration/5/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/5/direction: -1.0
 * /firefly/lee_position_controller_node/rotor_configuration/5/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/5/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/velocity_gain/x: 4.7
 * /firefly/lee_position_controller_node/velocity_gain/y: 4.7
 * /firefly/lee_position_controller_node/velocity_gain/z: 4.7
 * /firefly/robot_description: <?xml version="1....
 * /firefly/tf_prefix: 
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /use_sim_time: True

[... 49 lines cut ...]
[Dbg] [gazebo_motor_model.cpp:159] void gazebo::GazeboMotorModel::CreatePubsAndSubs() called.
[Dbg] [gazebo_motor_model.cpp:159] void gazebo::GazeboMotorModel::CreatePubsAndSubs() called.
[Dbg] [gazebo_motor_model.cpp:159] void gazebo::GazeboMotorModel::CreatePubsAndSubs() called.
[Dbg] [gazebo_motor_model.cpp:159] void gazebo::GazeboMotorModel::CreatePubsAndSubs() called.
[Dbg] [gazebo_motor_model.cpp:159] void gazebo::GazeboMotorModel::CreatePubsAndSubs() called.
[Dbg] [gazebo_motor_model.cpp:159] void gazebo::GazeboMotorModel::CreatePubsAndSubs() called.
[Dbg] [gazebo_controller_interface.cpp:106] CreatePubsAndSubs() called.
[Dbg] [gazebo_controller_interface.cpp:125] GazeboControllerInterface creating Gazebo publisher on "firefly/gazebo/command/motor_speed".
[Dbg] [gazebo_controller_interface.cpp:146] Subscribing to Gazebo topic "~/firefly/command/motor_speed".
[Dbg] [gazebo_controller_interface.cpp:165] CreatePubsAndSubs() called.
[... cut ...]
[Wrn] [Publisher.cc:132] Queue limit reached for topic /gazebo/default/broadcast_transform, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:132] Queue limit reached for topic /gazebo/default/firefly/joint_states, deleting message. This warning is printed only once.
[firefly/spawn_firefly-4] process has finished cleanly
log file: /home/jd/.ros/log/38c6bb74-39a4-11e7-b4b3-e4a7a00064b9/firefly-spawn_firefly-4*.log
[ INFO] [1494876322.290410835, 5.184000000]: Publishing waypoint on namespace //firefly: [0.000000, 0.000000, 1.000000].
`

JD-ETH, 2017-05-15:
After some comparison, I found out that the gazebo-plugin was not included in any other world file. 
By adding 
`    <plugin name="ros_interface_plugin" filename="librotors_gazebo_ros_interface_plugin.so"></plugin>`
This problem is solved