ethz-asl/rotors_simulator #360: rostors_simulators install

AlexMalo, 2017-03-17:
In the third step of the install procedure, when I execute any of the git clone commands I get

git clone git@github.com:ethz-asl/rotor_simulators.git
Cloning into 'rotor_simulators'...
The authenticity of host 'github.com (192.30.253.113)' can't be established.
RSA key fingerprint is SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8.
Are you sure you want to continue connecting (yes/no)? y
Please type 'yes' or 'no': yes
Warning: Permanently added 'github.com,192.30.253.113' (RSA) to the list of known hosts.
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

and yet I can download the files with the clone/download button.

pvechersky, 2017-03-17:
@AlexMalo This seems to be an HTTPS vs SSH issue. The git clone commands in step 3 of installation procedure use SSH URLs and require your SSH public key to be added to your Github account (https://help.github.com/articles/connecting-to-github-with-ssh/) ... Or do you have an SSH key set up already?

When you used the clone/download button did you use the HTTPS option?

@ffurrer Should we maybe have both SSH and HTTPS links in the instructions?

AlexMalo, 2017-03-17:
Ups, I am aware but not familiar with the public/private key usage. Sorry.
I copy the link and add it to git clone https://github.com/ethz-asl/rotors_simulator.git or I use the download the zip file button.

robrother, 2017-10-28:
Hi, my name is Erick, I'm trying to install RotorS and I was following the steps to install using Kinetic, but I have an error when I write _wstool merge rotors_hill.rosinstall_, could anybody help me to resolve this?
[image]

aeudes, 2017-10-28:
Hi, you should use this file : https://raw.githubusercontent.com/ethz-asl/rotors_simulator/master/rotors_hil.rosinstall see #443

robrother, 2017-10-28:
Hello @aeudes thanks for your soon answer, could you help me? I don´t know how install it, 
[image]

Jaeyoung-Lim, 2021-01-23:
Closing, since this seems to be related to wstool usage and not an issue of this repo

closed 2021-01-23 by Jaeyoung-Lim