ethz-asl/rotors_simulator #362: CAN the ROTOR S Be installed in Ubuntu 16.04 and ROS Kinetic?

shravank1996, 2017-03-18:
I tried installing Rotor S in Kinect but the system states the following:
ssh: connect to host github.com port 22: No route to host
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists

How Do I solve the problem?
[image]

vanurag, 2017-03-18:
@shravank1996 You are cloning with SSH option which requires you to have SSH keys. Try https version. so simply: git clone https://github.com/ethz-asl/rotors_simulator.git

shravank1996, 2017-03-19:
What about the other keys?

pvechersky, 2017-03-19:
@shravank1996 To fix this for the other repos, you can:

1) Set up the SSH connection to your github account (https://help.github.com/articles/connecting-to-github-with-ssh/). This way, you won't have to worry about this going forward.

OR

2) For each repo you want to clone, find it on Github with a search, click on 'Clone or download', make sure it says 'Clone with HTTPS' and copy the link it provides.

OR

3) For each repo listed in step 3 of installation instruction, just generate the HTTPS link yourself: replace "git@" with "https://" and the colon with a forward slash.

shravank1996, 2017-03-19:
Thanks or the help ,OK I am now getting a new error
Source space `/home/shravan/src` does
not yet exist.
 How do i rectify it?

shravank1996, 2017-03-19:
actually it exists in a different directory actually in windows folder

shravank1996, 2017-03-19:
Hey I got the system to work The Catkin Folder was not created Properly but then this error popped up when I try to run the system
Any help?
[image]

pvechersky, 2017-03-19:
- When you say that the catkin folder was not created properly what do you mean by that?
- Did the workspace build successfully?
- If the build did complete fine, it looks like ROS can't find your packages. Are you sourcing the .../devel/setup.bash file in your workspace?

shravank1996, 2017-03-20:
The Devel folder was not created Hence the system couldnt create Bash.setup.
Then when I reintialised catkin the workspace did build cause Bash did get set.
I did source it

shravank1996, 2017-03-29:
I think it was intialised properly and I did source the devel/setup.bash file into my workspace but now the system is showing a new error.
Can anyone please help me
[image]

shravank1996, 2017-03-29:
[image]
Terminal says Broken Gazebo Package
Is it a problem caused by me having gazebo installed before installing ROS?
Please help

robrother, 2017-10-28:
Hi, my name is Erick, I'm trying to install RotorS and I was following the steps to install using Kinetic, but I have an error when I write _wstool merge rotors_hill.rosinstall_, could anybody help me to resolve this?
[image]

ffurrer, 2017-11-17:
Please open new issues if you have specific questions, will close this one.

closed 2017-11-17 by ffurrer