ethz-asl/rotors_simulator #367: ros kinetic hil_interface mavlink include

jgoppert, 2017-03-20:
```bash
Errors     << rotors_hil_interface:make /home/jgoppert/git/catkin/logs/rotors_hil_interface/build.make.000.log
In file included from /home/jgoppert/git/catkin/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:17:0:
/home/jgoppert/git/catkin/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:105:3: error: ‘mavlink_hil_gps_t’ does not name a type
```

I am working on fixing this. If I include the mavlink common header I get a bit further but get more errors. @gbmhunter have you run into this before?

gbmhunter, 2017-03-21:
@jgoppert I have definitely received this error before, I can't remember exactly when and why though sorry! I remember "grepping" the type name through my entire computer and finding the right include through that.

taylorbodin, 2017-05-14:
@jgoppert Please let me know if you figure out what to include to get this to build properly. I am running into the same issues.

Maunsi, 2017-08-14:
Hello, has anyone found the solution to this?

msadowski, 2017-08-17:
I'm running into the same issue. The hil messages seem to be defined in [mavlink generated lib](https://github.com/mavlink/c_library_v1/tree/master/common) (which doesn't seem to be included in any of the packages specified in the [HIL docs](https://github.com/ethz-asl/rotors_simulator/wiki/Setting-up-RotorS-as-Fixed-Wing-HiL-Simulation-(Pixhawk,-Mavros,-Mavlink,-QGC))).

zarmomin, 2017-09-29:
What solved it for me was to follow this instruction on the proper installation of mavros https://dev.px4.io/en/setup/dev_env_linux_ubuntu.html

if you still get an import error ("model future missing"), `pip install future`

v01d, 2018-05-09:
Any workaround for this? The full error on kinetic is:

<code>
[ 20%] Building CXX object CMakeFiles/rotors_hil_interface.dir/src/hil_sensor_level_interface.cpp.o         
In file included from /home/v01d/ros/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:17:0:
/home/v01d/ros/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:105:3: error: ‘mavlink_hil_gps_t’ does not name a type
   mavlink_hil_gps_t hil_gps_msg_;                                                                          
   ^                                                                                                        
/home/v01d/ros/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:108:3: error: ‘mavlink_hil_sensor_t’ does not name a type
   mavlink_hil_sensor_t hil_sensor_msg_;                                                                    
   ^                                                                                                        
/home/v01d/ros/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:130:3: error: ‘mavlink_hil_state_quaternion_t’ does not name a type
   mavlink_hil_state_quaternion_t hil_state_qtrn_msg_;                                                      
   ^                                                                                                        
/home/v01d/ros/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp: In member function ‘virtual std::vector<mavros_msgs::Mavlink_<std::allocator<void> > > rotors_hil::HilSensorLevelInterface::CollectData()’:
/home/v01d/ros/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:89:3: error: ‘mavlink_message_t’ was not declared in this scope
   mavlink_message_t mmsg;                                                                                  
   ^                                                                                                        
/home/v01d/ros/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:89:3: note: suggested alternatives:
In file included from /opt/ros/kinetic/include/mavlink/v2.0/common/../message.hpp:15:0,                     
                 from /opt/ros/kinetic/include/mavlink/v2.0/common/common.hpp:16,                           
                 from /opt/ros/kinetic/include/mavconn/mavlink_dialect.h:26,                                
                 from /opt/ros/kinetic/include/mavros_msgs/mavlink_convert.h:18,                            
                 from /home/v01d/ros/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:23,
                 from /home/v01d/ros/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:17:
/opt/ros/kinetic/include/mavlink/v2.0/common/../mavlink_types.h:121:4: note:   ‘mavlink::mavlink_message_t’ 
 }) mavlink_message_t;                                                                                      
    ^                                                                                                       
/opt/ros/kinetic/include/mavlink/v2.0/common/../mavlink_types.h:121:4: note:   ‘mavlink::mavlink_message_t’ 
/home/v01d/ros/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:105:5: error: ‘hil_gps_msg_’ was not declared in this scope
     hil_gps_msg_.time_usec = time_usec;                                                                    
     ^                                                                                                      
/home/v01d/ros/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:119:5: error: ‘mavlink_hil_gps_t’ was not declared in this scope
     mavlink_hil_gps_t* hil_gps_msg_ptr = &hil_gps_msg_;                                                    
     ^                                                                                                      
/home/v01d/ros/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:119:24: error: ‘hil_gps_msg_ptr’ was not declared in this scope
     mavlink_hil_gps_t* hil_gps_msg_ptr = &hil_gps_msg_;                                                    
                        ^                                                                                   
/home/v01d/ros/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:120:39: error: ‘mmsg’ was not declared in this scope
     mavlink_msg_hil_gps_encode(1, 0, &mmsg, hil_gps_msg_ptr);                                              
                                       ^                                                                    
/home/v01d/ros/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:120:60: error: ‘mavlink_msg_hil_gps_encode’ was not declared in this scope
     mavlink_msg_hil_gps_encode(1, 0, &mmsg, hil_gps_msg_ptr);                                              
                                                            ^                                               
/home/v01d/ros/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:131:3: error: ‘hil_sensor_msg_’ was not declared in this scope
   hil_sensor_msg_.time_usec = time_usec;                                                                   
[... cut ...]
[build] Summary: 1 of 2 packages succeeded.                                                                
[build]   Ignored:   44 packages were skipped or are blacklisted.                                          
[build]   Warnings:  None.                                                                                 
[build]   Abandoned: None.                                                                                 
[build]   Failed:    1 packages failed.                                                                    
[build] Runtime: 11.0 seconds total. 
</code>

Jaeyoung-Lim, 2021-01-24:
This should be fixed now, closing

closed 2021-01-24 by Jaeyoung-Lim