ethz-asl/rotors_simulator #373: Problem with mav_with_waypoint_publisher.launch example

zgqball, 2017-03-22:
i want to use waypoint_publisher to plan a path for mav.
BUT when i  tried to launch the mav_with_waypoint_publisher.launch , everything was fine except the waypoint_publisher_file, which was automatically  cleaned.WHY and how to solve it?

here is the summary.

SUMMARY
========

PARAMETERS
 * /firefly/lee_position_controller_node/angular_rate_gain/x: 0.52
 * /firefly/lee_position_controller_node/angular_rate_gain/y: 0.52
 * /firefly/lee_position_controller_node/angular_rate_gain/z: 0.18
 * /firefly/lee_position_controller_node/attitude_gain/x: 3
 * /firefly/lee_position_controller_node/attitude_gain/y: 3
 * /firefly/lee_position_controller_node/attitude_gain/z: 0.15
 * /firefly/lee_position_controller_node/inertia/xx: 0.0347563
 * /firefly/lee_position_controller_node/inertia/xy: 0.0
 * /firefly/lee_position_controller_node/inertia/xz: 0.0
 * /firefly/lee_position_controller_node/inertia/yy: 0.0458929
 * /firefly/lee_position_controller_node/inertia/yz: 0.0
 * /firefly/lee_position_controller_node/inertia/zz: 0.0977
 * /firefly/lee_position_controller_node/mass: 1.56779
 * /firefly/lee_position_controller_node/position_gain/x: 6
 * /firefly/lee_position_controller_node/position_gain/y: 6
 * /firefly/lee_position_controller_node/position_gain/z: 6
 * /firefly/lee_position_controller_node/rotor_configuration/0/angle: 0.52359877559
 * /firefly/lee_position_controller_node/rotor_configuration/0/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/0/direction: 1.0
 * /firefly/lee_position_controller_node/rotor_configuration/0/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/0/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/1/angle: 1.57079632679
 * /firefly/lee_position_controller_node/rotor_configuration/1/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/1/direction: -1.0
 * /firefly/lee_position_controller_node/rotor_configuration/1/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/1/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/2/angle: 2.61799387799
 * /firefly/lee_position_controller_node/rotor_configuration/2/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/2/direction: 1.0
 * /firefly/lee_position_controller_node/rotor_configuration/2/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/2/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/3/angle: -2.61799387799
 * /firefly/lee_position_controller_node/rotor_configuration/3/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/3/direction: -1.0
 * /firefly/lee_position_controller_node/rotor_configuration/3/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/3/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/4/angle: -1.57079632679
 * /firefly/lee_position_controller_node/rotor_configuration/4/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/4/direction: 1.0
 * /firefly/lee_position_controller_node/rotor_configuration/4/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/4/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/rotor_configuration/5/angle: -0.52359877559
 * /firefly/lee_position_controller_node/rotor_configuration/5/arm_length: 0.215
 * /firefly/lee_position_controller_node/rotor_configuration/5/direction: -1.0
 * /firefly/lee_position_controller_node/rotor_configuration/5/rotor_force_constant: 8.54858e-06
 * /firefly/lee_position_controller_node/rotor_configuration/5/rotor_moment_constant: 0.016
 * /firefly/lee_position_controller_node/velocity_gain/x: 4.7
 * /firefly/lee_position_controller_node/velocity_gain/y: 4.7
 * /firefly/lee_position_controller_node/velocity_gain/z: 4.7
 * /firefly/robot_description: <?xml version="1....
 * /firefly/tf_prefix: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.6
 * /use_sim_time: True

NODES
  /firefly/
    hovering_example (rotors_gazebo/hovering_example)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    lee_position_controller_node (rotors_control/lee_position_controller_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_firefly (gazebo_ros/spawn_model)
    waypoint_publisher_file (rotors_gazebo/waypoint_publisher_file)
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)

auto-starting new master
process[master]: started with pid [9960]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f9e089a4-0eaf-11e7-a5ad-e839dfdbba03
process[rosout-1]: started with pid [9973]
started core service [/rosout]
process[gazebo-2]: started with pid [9997]
process[gazebo_gui-3]: started with pid [10002]
process[firefly/spawn_firefly-4]: started with pid [10007]
process[firefly/lee_position_controller_node-5]: started with pid [10008]
process[firefly/hovering_example-6]: started with pid [10011]
process[firefly/waypoint_publisher_file-7]: started with pid [10039]
[ INFO] [1490153466.601706907]: Started hovering example.
[ INFO] [1490153466.605113502]: Wait for 1 second before trying to unpause Gazebo again.
process[firefly/robot_state_publisher-8]: started with pid [10052]
process[firefly/joint_state_publisher-9]: started with pid [10058]
[ INFO] [1490153466.716463539]: Started waypoint_publisher.
[ INFO] [1490153466.716711746]: Read 4 waypoints.
[ INFO] [1490153466.775819158]: Wait for simulation to become ready...
SpawnModel script started
[INFO] [1490153467.228502, 0.000000]: Loading model XML from ros parameter
[INFO] [1490153467.235381, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[ INFO] [1490153467.497426884]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1490153467.499148606]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1490153467.606604383]: Wait for 1 second before trying to unpause Gazebo again.
[INFO] [1490153468.445487, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1490153468.928784, 0.000000]: Spawn status: SpawnModel: Successfully spawned model
[ INFO] [1490153468.953360285]: Unpaused the Gazebo simulation.
[firefly/spawn_firefly-4] process has finished cleanly
log file: /home/adam/.ros/log/f9e089a4-0eaf-11e7-a5ad-e839dfdbba03/firefly-spawn_firefly-4*.log
[ INFO] [1490153469.225005508, 0.030000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1490153469.326907172, 0.130000000]: Physics dynamic reconfigure ready.
[ INFO] [1490153469.502064645, 0.310000000]: ...ok
[ INFO] [1490153474.229114024, 5.010000000]: Publishing waypoint on namespace //firefly: [0.000000, 0.000000, 1.000000].
[ INFO] [1490153499.550396504, 30.310000000]: Start publishing waypoints.
**[firefly/waypoint_publisher_file-7] process has finished cleanly
log file: /home/adam/.ros/log/f9e089a4-0eaf-11e7-a5ad-e839dfdbba03/firefly-waypoint_publisher_file-7*.log**

many thanks

Michael9792, 2017-03-23:
Hi, when I run `roslaunch rotors_gazebo mav_with_waypoint_publisher.launch mav_name:=firefly world_name:=basic
`command, I got the same result as yours. 

However, I'm quite sure `[firefly/waypoint_publisher_file-7] process has finished cleanly` is not a problem. If you run `mav_hovering_example` node or `waypoint_publisher` node, you will find both nodes clean their process after they publish the waypoints. 
So I guess there is something wrong in my usage of code.

If you really want to send several waypoints in order, my temporary solution is to write a new node or modify hovering_example.cpp, making it publish several waypoints one by one. But this is not really a good idea. I think there should be more effective ways to solve it.

zgqball, 2017-03-27:
@Michael9792 ,thanks for ur help! It does work eventually!

btw, do u know how to get the position for the mav? I want to generate a txt file of these position.

Michael9792, 2017-04-05:
@zgqball One way is to use `rosbag` tools provided by ROS itself. You could record the data or even replay all the data. It's very useful.
Another idea is that you could write a new node in your catkin workspace. The node do nothing but subscribes mav position messages. After that, you could use whatever ways to write the data to files.

ffurrer, 2017-04-05:
You can also take a look at our bag plugin (https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo_plugins/src/gazebo_bag_plugin.cpp), which essentially subscribes to topics and writes them to a bag file, you could just write to a local file instead. @zgqball for the next time, if you have questions, or issues that are unrelated to this issue, just open a new one and close the one which is resolved.

zgqball, 2017-05-19:
@ffurrer ,thanks for the advices above. And, could u tell me how to use the bag plugin? i cannot find out the example of bag plugin.

closed 2017-06-14 by timethy

FanJing1112, 2021-04-12:
Hello. I want to ask how u solved it? Please.

PR #406 (2017-06-14): FIX publish msg and not msg pointer in waypoint_publisher
Waypoint_publisher didn't work for me, so I dug into it.

Seems like the problem was that the pointer to the message was published instead of the message itself.
 
This would fix #373 I think.