ethz-asl/rotors_simulator #380: Update Mavros tests with usage of rotors_simulator

TSC21, 2017-04-05:
Related to https://github.com/mavlink/mavros/issues/506. The idea is to sync up the tests with your models, since it seems that the PX4 SITL ROS/Gazebo has been left discontinued. Will keep you updated

TSC21, 2017-04-05:
Also suggestions and tips are welcomed, if the dev team requires those. Thanks

TSC21, 2017-04-06:
@ffurrer, @pvechersky ping

ffurrer, 2017-04-06:
@TSC21 Great to hear, that you are syncing with us! I'm including @hitimo in this, as I'm haven't worked with mavlink, so he should be the right person to talk to.

TSC21, 2017-04-06:
Yeah I saw the need of doing it as in the PX4 Firmware they are leaving ROS Gazebo SITL aside. So I think is best to sync up with you since you have things more clear on the ROS bridge. Also, together with this, I'm improving the HIL bridge with MAVROS so to allow to use your package with PX4 HIL in a more cleaner way. Related is https://github.com/mavlink/mavros/pull/684 and #356.

pvechersky, 2017-04-09:
@TSC21 Thank you for looking into this! The changes in [mavlink/mavros#684](https://github.com/mavlink/mavros/pull/684) look really useful and helping simplify our rotors_hil_interface package.

My one question after looking at that pull request, however, is why are the hil_sensor messages throttled to 40 Hz? Is there a reason for that specific number? ... At least in case of PX4, I was under the impression that certain modules, like the state estimator, expect/operate better with the IMU measurements provided at roughly 100 Hz, no?

TSC21, 2017-04-09:
Well that's a good question and something I retrieved from a previous commit. @mzahana can you explain why you throttled this?

TSC21, 2017-04-09:
> @TSC21 Thank you for looking into this! The changes in mavlink/mavros#684 look really useful and helping simplify our rotors_hil_interface package.

I'll PR the changes on your HIL interface after this is merged, so we are also able to use kinetic version of mavros with your package (since it was giving building errors on the `hil_interface` since it was depending on the mavros older structure from `indigo-devel`).

pvechersky, 2017-04-09:
Sounds good! Thank you for adapting our HIL interface package to your changes once they're merged :)

mzahana, 2017-04-09:
That was long time ago when I was having some weird issues with the stability, and I was having doubts that sensors data need to be at specific rate. So I was trying different rates.

This, however, should be re-set/removed. There is no need for that throttle. It still needs testing , though.