ethz-asl/rotors_simulator #381: Wont' build on ROS kinetic due to OGRE header path

jmwest0774, 2017-04-07:
get the following error compiling on Ubunut 16.04/Kinetic

In file included from /usr/include/gazebo-7/gazebo/rendering/ogre_gazebo.h:61:0,
                 from /usr/include/gazebo-7/gazebo/rendering/Camera.hh:50,
                 from /home/nathan/temp_ws/src/rotors_simulator/rotors_gazebo_plugins/include/rotors_gazebo_plugins/external/gazebo_geotagged_images_plugin.h:25,
                 from /home/nathan/temp_ws/src/rotors_simulator/rotors_gazebo_plugins/src/external/gazebo_geotagged_images_plugin.cpp:18:
/usr/include/OGRE/Terrain/OgreTerrainPaging.h:33:35: fatal error: OgrePagedWorldSection.h: No such file or directory
compilation terminated.

Fixed by changing line 48 of rotor_gazebo_plugins/CMakeLists.txt to:
include_directories(${ADDITIONAL_INCLUDE_DIRS} ${OGRE_INCLUDE_DIRS}/Paging)

lecampos, 2017-07-05:
Same issue, same fix. 

thanks =)

Unity-W, 2019-05-28:
Thanks +1

SaimonMR, 2019-10-09:
Thanks so much, problem fixed!

Rinkal2, 2019-11-11:
bhaiiiiiii bhaiiiiii bhaiiiiiiiiiii

BIRL-xu, 2021-02-27:
Fixed the issue! Thanks!

JiangxinkeTao, 2022-08-06:
Fixed this issues Thanks