ethz-asl/rotors_simulator #387: Quadrotors sharing paylod

PedroHRPBS, 2017-04-10:


ffurrer, 2017-04-18:
@fmina has worked on similar problems.
So just pinging him here, he might be able to give you some hints.

fmina, 2017-04-18:
To create a spherical joint you will have to combine three revolute joints. 
Controlling two MAVs doesn't need any modification of the plugins. You just have to use correct namespace for each MAV, for example `firefly1` and `firefly2`.

PedroHRPBS, 2017-04-18:
But then, how do I add the bar to connect them? Changing the namespace is done on the .launch file, right? Is it possible to create joints there?

fmina, 2017-04-27:
No, two MAVs connected with one object is seen by gazebo as single robot. You need to prepare a urdf file that creates the mechanics of two MAVs and connect them with joints and links. The MAV mechanics is defined here: https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_description/urdf/firefly.xacro

PedroHRPBS, 2017-04-29:
So, to have a "robot" that constains 2 quads, I would have to instantiate 2 quads inside of this file that you sent? I want that the two quads fly independently.

juanmed, 2019-06-27:
@PedroHRPBS  Hi. I see this thread has been around for a long time. Did you by any chance get to work this around? Joining two quads with a bar? If yes, would you mind pointing to any examples on how you got it working? Thanks!

PedroHRPBS, 2019-06-27:
@juanmed actually I gave up, this was a project that I was working on at my graduation. But it took another path, so I didn't end up solving this. Sorry.

juanmed, 2019-06-27:
I see. Thanks for your answer!