ethz-asl/rotors_simulator #388: Catkin build error

SaiManikanta23, 2017-04-15:
**_Hello I've followed the steps provided, but instead of indigo I've used ros-kinectic, so this is the issue i got with the catkin build, is this because that I am using kinetic instead of indigo??_**
```

bash: cd: catkin_ws: No such file or directory
sai@sai-VirtualBox:~/catkin_ws$ rm -rf build/ devel/
sai@sai-VirtualBox:~/catkin_ws$ catkin build
--------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /home/sai/catkin_ws
--------------------------------------------------------
Source Space:       [exists] /home/sai/catkin_ws/src
Log Space:          [exists] /home/sai/catkin_ws/logs
Build Space:        [exists] /home/sai/catkin_ws/build
Devel Space:        [exists] /home/sai/catkin_ws/devel
Install Space:      [unused] /home/sai/catkin_ws/install
DESTDIR:            [unused] None
--------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------
[build] Found '12' packages in 0.0 seconds.                                    
[build] Updating package table.                                                
Warning: generated devel space setup files have been deleted.
Starting  >>> catkin_tools_prebuild                                            
_______________________________________________________________________________
Errors     << catkin_tools_prebuild:cmake /home/sai/catkin_ws/logs/catkin_tools_prebuild/build.cmake.003.log
CMake Error at /opt/ros/kinetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/local/bin/python
  "/opt/ros/kinetic/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/kinetic/share/catkin/cmake/../package.xml"
  "/home/sai/catkin_ws/build/catkin_tools_prebuild/catkin/catkin_generated/version/package.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package_xml.cmake:63 (safe_execute_process)
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:151 (_catkin_package_xml)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

cd /home/sai/catkin_ws/build/catkin_tools_prebuild; catkin build --get-env catkin_tools_prebuild | catkin env -si  /usr/bin/cmake /home/sai/catkin_ws/build/catkin_tools_prebuild --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/sai/catkin_ws/devel/.private/catkin_tools_prebuild -DCMAKE_INSTALL_PREFIX=/home/sai/catkin_ws/install; cd -
...............................................................................
Failed     << catkin_tools_prebuild:cmake          [ Exited with code 1 ]      
Failed    <<< catkin_tools_prebuild                [ 2.6 seconds ]             
Abandoned <<< mav_msgs                             [ Unrelated job failed ]    
Abandoned <<< planning_msgs                        [ Unrelated job failed ]    
Abandoned <<< rotors_comm                          [ Unrelated job failed ]    
Abandoned <<< rotors_control                       [ Unrelated job failed ]    
Abandoned <<< rotors_description                   [ Unrelated job failed ]    
Abandoned <<< rotors_evaluation                    [ Unrelated job failed ]    
Abandoned <<< rotors_gazebo_plugins                [ Unrelated job failed ]    
Abandoned <<< rotors_gazebo                        [ Unrelated job failed ]    
Abandoned <<< rotors_hil_interface                 [ Unrelated job failed ]    
Abandoned <<< rotors_joy_interface                 [ Unrelated job failed ]    
[build] Summary: 0 of 11 packages succeeded.                                   
[build]   Ignored:   2 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 10 packages were abandoned.                               
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 2.7 seconds total. 
```

ffurrer, 2017-05-03:
@SaiManikanta23 did you resolve your issue?

kinetic shouldn't be a problem.
It's hard from the output to figure out what the problem is, also the screenshot and the output above are not the same, maybe you should try again from a clean catkin workspace.

SaiManikanta23, 2017-05-03:
Hello

Thank you for the reply and Yes, I have re installed everything from scratch and everything seems to be working now. There is a problem with installation path of Catkin and that caused the issue

Thanks
Sai

closed 2017-05-03 by SaiManikanta23