ethz-asl/rotors_simulator #390: Failing to bulid on Indigo

tfoote, 2017-04-22:
The indigo build of  gazebo_rotors_plugins is failing after the last 3.2.0 release. 
http://build.ros.org/view/Ibin_uT64/job/Ibin_uT64__rotors_gazebo_plugins__ubuntu_trusty_amd64__binary/39/console

The console output suggests there may be a CMake error with the wrong path. 

Also filed at https://github.com/ethz-asl/mav_comm/issues/52 as https://github.com/ros/rosdistro/pull/14488 appeared to trigger the first failing build

```
23:48:41 -- Gazebo version is less than 5, not building gazebo_lidar_plugin.cpp.
23:48:41 -- BUILD_MAVLINK_INTERFACE_PLUGIN = FALSE, NOT adding mavros dependency and NOT building mavlink_interface_plugin.
23:48:41 -- Gazebo version is less than 5, not building liftdrag_plugin.cpp.
23:48:41 -- CMAKE_INSTALL_PREFIX = /opt/ros/indigo
23:48:56 CMake Error at CMakeLists.txt:459 (file):
23:48:56   file COPY cannot set permissions on
23:48:56   "/tmp/binarydeb/ros-indigo-rotors-gazebo-plugins-2.1.0/obj-x86_64-linux-gnu/devel/include//devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/devel/include/gtest/CMakeFiles"
23:48:56 
23:48:56 
```

tfoote, 2017-04-26:
We're planning an indigo sync soon: https://discourse.ros.org/t/indigo-sync-preparations-2017-04-22/1709/1 if this is not resolved the package will not be available due to this regression if we can't fix it shortly.

ffurrer, 2017-04-27:
Closing this, as I think all the packages are building fine again. @tfoote please reopen, if this is not the case.
Cheers

closed 2017-04-27 by ffurrer