ethz-asl/rotors_simulator #399: CMakeList PROTOBUF compilation error

lecampos, 2017-05-09:
The error was solved by changing the line 140 of the CMakeList.txt file in the rotors_gazebo_plugins package for set(PROTOBUF_IMPORT_DIRS ".")

fmina, 2017-05-10:
Can you please write the error message?

marija-p, 2017-05-12:
Hi,

I had the exact same issue when trying to build RotorS on my Ubuntu 14.04 today. I used the same fix. The problem package was ```rotors_gazebo_plugins```.

Error message:
```
Errors     << rotors_gazebo_plugins:make /home/marija/catkin_ws/logs/rotors_gazebo_plugins/build.make.005.log
make[2]: PROTOBUF_PROTOC_EXECUTABLE-NOTFOUND: Command not found
```

mandu46, 2017-06-20:
Hi, 
It looks like my problem is same as this issue, when I try to build RotorS on my Ubuntu 14.04 todaym I have bellow messages. I can`t do the catkin build ...

[rotors_gazebo_plugins] ==> '/home/drone0/sim_ws/build/rotors_gazebo_plugins/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/drone0/sim_ws/build/rotors_gazebo_plugins'
make[2]: *** No rule to make target `/home/drone0/sim_ws/src/rotors_simulator/rotors_gazebo_plugins/PROTOBUF_PROTOC_EXECUTABLE-NOTFOUND', needed by `WindSpeed.pb.cc'.  Stop.

marija-p, 2017-06-20:
Hey @mandu46 , I had this problem again yesterday too! This time I solved it by running:

``
sudo apt-get install protobuf-compiler
cd catkin_ws
rm -rf build/ devel/
catkin build 
``

as discussed in Issue #354 . I hope this fix works for you as well.

marija-p, 2017-06-20:
@fmina Maybe we should write the fix down somewhere more obvious as it seems to be a very common issue.

mandu46, 2017-06-20:
Hi @marija-p . It works!! Thank you for your help! I really appreciate for it !!

ffurrer, 2017-06-20:
@marija-p added the install command here: https://github.com/ethz-asl/rotors_simulator/wiki/Setting-up-the-RotorS-Simulator
Thanks :)

closed 2017-06-20 by ffurrer