ethz-asl/rotors_simulator #400: Tricopter with tiltrotor - gazebo_multirotors_base_plugin

ganeshrk95, 2017-05-11:
Hey guys!

I'm working on Tricopter with a tilt serv operating the back motor. I have not seen any model with actuated links in series. In my model, the parent-chile tree is as follows:

->base_link
----->Front_right
----->Front_left
----->TILT_servo
----------->Back_motor

All the reference models/examples provided have a single base_link (parent) and rotors as child links. In my case the TILT_servo is the parent for Back_motor. Can someone guide how to go about this ?

Thanks in advance.

-Ganesh

ganeshrk95, 2017-06-15:
I am using the PX4 stack. When I command the servo link to take some position, I cannot get the back motor to rotate which is on another channel. I have setup the mavlink_controller in the following manner:
Ch1: Front Right Motor
Ch2: Front Left Motor
Ch3: Back Motor
Ch4: Servo Motor

I can give only one. either Ch3 or Ch4. I think the problem arises because BACK MOTOR's parent link is the SERVO MOTOR and the SERVO MOTORs' parent is the BASE_LINK.

ganeshrk95, 2017-06-21:
If you can guide me regarding the part of the code which needs to be modified, that would be of great help.

ghostold, 2017-08-27:
did U get it worked？could U please share the way to setup the back servo？

suraj2596, 2018-12-15:
@ganeshrk95 , could you elaborate on your project little more?

Jaeyoung-Lim, 2021-01-24:
Closing stale

closed 2021-01-24 by Jaeyoung-Lim