ethz-asl/rotors_simulator #401: How to import two hummingbird uavs in one scene?

jiaweimeng, 2017-05-15:
Hi guys,

Recently I am using gazebo and working on uavs simulation. What I plan to do is use multi-uavs to do some cool path planning. However, the issue now is that I cannot import 2 hummingbird uavs in one scene and the error shows that the model name "hummingbird" has already exist. Can anyone help me with this problem please? 

Here is my code:
<launch>
  <arg name="world_name" default="basic"/>
  <arg name="enable_logging" default="false"/>
  <arg name="enable_ground_truth" default="true"/>
  <arg name="paused" value="false"/>

  <env name="GAZEBO_MODEL_PATH" value="${GAZEBO_MODEL_PATH}:$(find rotors_gazebo)/models"/>
  <env name="GAZEBO_RESOURCE_PATH" value="${GAZEBO_RESOURCE_PATH}:$(find rotors_gazebo)/models"/>
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find rotors_gazebo)/worlds/$(arg world_name).world"/>
    
    <arg name="paused" value="$(arg paused)"/>
    
  </include>

  <group ns="hummingbird">
    <include file="$(find rotors_gazebo)/launch/spawn_mav.launch">
      <arg name="mav_name" value="hummingbird" />
      <arg name="model" value="$(find rotors_description)/urdf/mav_generic_odometry_sensor.gazebo" />
      <arg name="enable_logging" value="$(arg enable_logging)" />
      <arg name="enable_ground_truth" value="$(arg enable_ground_truth)" />
      <arg name="log_file" value="hummingbird"/>
      <arg name="y" value="0.0"/>
    </include>
    <node name="lee_position_controller_node" pkg="rotors_control" type="lee_position_controller_node" output="screen">
      <rosparam command="load" file="$(find rotors_gazebo)/resource/lee_controller_hummingbird.yaml" />
      <rosparam command="load" file="$(find rotors_gazebo)/resource/hummingbird.yaml" />
      <remap from="odometry" to="odometry_sensor1/odometry" />
    </node>
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
    <node name="waypoint_publisher_file" pkg="rotors_gazebo" type="waypoint_publisher_file" output="screen" args="$(find rotors_gazebo)/resource/example_waypoints2.txt"/>
  </group>

  <group ns="hummingbird_1">
    <include file="$(find rotors_gazebo)/launch/spawn_mav.launch">
      <arg name="mav_name" value="hummingbird" />
      <arg name="model" value="$(find rotors_description)/urdf/mav_generic_odometry_sensor.gazebo" />
      <arg name="enable_logging" value="$(arg enable_logging)" />
      <arg name="enable_ground_truth" value="$(arg enable_ground_truth)" />
      <arg name="log_file" value="hummingbird"/>
      <arg name="y" value="1.0"/>
    </include>
    <node name="lee_position_controller_node" pkg="rotors_control" type="lee_position_controller_node" output="screen">
      <rosparam command="load" file="$(find rotors_gazebo)/resource/lee_controller_hummingbird.yaml" />
      <rosparam command="load" file="$(find rotors_gazebo)/resource/hummingbird.yaml" />
      <remap from="odometry" to="odometry_sensor1/odometry" />
    </node>
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
    <node name="waypoint_publisher_file" pkg="rotors_gazebo" type="waypoint_publisher_file" output="screen" args="$(find rotors_gazebo)/resource/example_waypoints1.txt"/>
  </group>

  <group ns="firefly">
    <include file="$(find rotors_gazebo)/launch/spawn_mav.launch">
      <arg name="mav_name" value="firefly" />
      <arg name="model" value="$(find rotors_description)/urdf/mav_generic_odometry_sensor.gazebo" />
      <arg name="enable_logging" value="$(arg enable_logging)" />
      <arg name="enable_ground_truth" value="$(arg enable_ground_truth)" />
      <arg name="log_file" value="firefly"/>
      <arg name="y" value="2.0"/>
    </include>
    <node name="lee_position_controller_node" pkg="rotors_control" type="lee_position_controller_node" output="screen">
      <rosparam command="load" file="$(find rotors_gazebo)/resource/lee_controller_firefly.yaml" />
      <rosparam command="load" file="$(find rotors_gazebo)/resource/firefly.yaml" />
      <remap from="odometry" to="odometry_sensor1/odometry" />
    </node>
    <node name="waypoint_publisher_file" pkg="rotors_gazebo" type="waypoint_publisher_file" output="screen" args="$(find rotors_gazebo)/resource/example_waypoints.txt"/>
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
  </group>
</launch>

Thank you very much:)

jiaweimeng, 2017-05-15:
<group ns="hummingbird">
    <include file="$(find rotors_gazebo)/launch/spawn_mav.launch">
      <arg name="mav_name" value="hummingbird" />
      <arg name="model" value="$(find rotors_description)/urdf/mav_generic_odometry_sensor.gazebo" />
      <arg name="enable_logging" value="$(arg enable_logging)" />
      <arg name="enable_ground_truth" value="$(arg enable_ground_truth)" />
      <arg name="log_file" value="hummingbird"/>
      <arg name="y" value="0.0"/>
    </include>
    <node name="lee_position_controller_node" pkg="rotors_control" type="lee_position_controller_node" output="screen">
      <rosparam command="load" file="$(find rotors_gazebo)/resource/lee_controller_hummingbird.yaml" />
      <rosparam command="load" file="$(find rotors_gazebo)/resource/hummingbird.yaml" />
      <remap from="odometry" to="odometry_sensor1/odometry" />
    </node>
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
    <node name="waypoint_publisher_file" pkg="rotors_gazebo" type="waypoint_publisher_file" output="screen" args="$(find rotors_gazebo)/resource/example_waypoints2.txt"/>
  </group>

  <group ns="hummingbird_1">
    <include file="$(find rotors_gazebo)/launch/spawn_mav.launch">
      <arg name="mav_name" value="hummingbird" />
      <arg name="model" value="$(find rotors_description)/urdf/mav_generic_odometry_sensor.gazebo" />
      <arg name="enable_logging" value="$(arg enable_logging)" />
      <arg name="enable_ground_truth" value="$(arg enable_ground_truth)" />
      <arg name="log_file" value="hummingbird"/>
      <arg name="y" value="1.0"/>
    </include>
    <node name="lee_position_controller_node" pkg="rotors_control" type="lee_position_controller_node" output="screen">
      <rosparam command="load" file="$(find rotors_gazebo)/resource/lee_controller_hummingbird.yaml" />
      <rosparam command="load" file="$(find rotors_gazebo)/resource/hummingbird.yaml" />
      <remap from="odometry" to="odometry_sensor1/odometry" />
    </node>
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
    <node name="waypoint_publisher_file" pkg="rotors_gazebo" type="waypoint_publisher_file" output="screen" args="$(find rotors_gazebo)/resource/example_waypoints1.txt"/>
  </group>

marco-tranzatto, 2017-05-15:
Check how groups and namespaces (documentation [here](http://wiki.ros.org/roslaunch/XML/group) ) are used in this launch file: [https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo/launch/firefly_swarm_hovering_example.launch](https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo/launch/firefly_swarm_hovering_example.launch)

;)

ffurrer, 2017-06-15:
@jiaweimeng if the above suggestion worked for you, can you close this?

closed 2017-06-17 by jiaweimeng