ethz-asl/rotors_simulator #419: virtual joystick is not working.

durgesh0071, 2017-06-30:
while running  mav_with_keyboard.launch file everything goes well. but when I change the throttle or any input on virtual joystick , I am not getting any output in gazebo simulation.

pvechersky, 2017-07-04:
Hi @durgesh0071,

Thank you for your feedback! ... I just tested 'mav_with_keyboard.launch' on my machine with Ubuntu 14.04 and it worked (although the throttle seems really sensitive - consider reducing the max thrust value if you get this working).

One thing I did notice is that this launch file starts Gazebo in paused state. Could this be the problem? Are you un-pausing the simulation before trying to fly with the virtual joystick?

If that's not the problem then another thing to try is to run add the 'verbose' flag to Gazebo when running this launch file to see it something else is failing - perhaps some plugin not being initialized properly, etc. ... This might give us more information as to what the problem might be ...

durgesh0071, 2017-07-04:
Hi,

Thanks for reply. yes I am un-pausing the simulation before trying to fly
with the virtual joystick. And I  tried to run gazebo with verbose flag by
using (rosparam set verbose true). But still I am getting the same problem.

I have attached the rqt_graphs with active node and all node related
to mav_with_keyboard.launch
file.

with regards,
Durgesh

pvechersky, 2017-07-05:
Looks like the attachment didn't make it :(

Can you also verify with rostopic echo that when you change the input on the virtual joystick GUI the 'joy' node publishes proper joystick commands? ... This way we can maybe see if the problem is on the RotorS plugins side or in the device communication ...

durgesh0071, 2017-07-05:
while running this 'rostopic echo /firefly/joy ,I am  getting a warning message-
WARNING: no messages received and simulated time is active.
Is /clock being published?
I checked gazebo simulation is in unpaused mode.
sorry for file not got attached ,below I have attached a link which containing the rqt_graphs.
https://drive.google.com/open?id=0BwulAwNS0fVzak14ak1nQmEtQ1U

pvechersky, 2017-07-09:
Sorry for the late response :(

Hmm, that's interesting, my rqt_graph looks different when I ran mav_with_keyboard.launch (see the attached image). Just to clarify - did you follow these instructions for setting up the virtual joystick? https://github.com/ethz-asl/rotors_simulator/wiki/Setup-virtual-keyboard-joystick ... Are you using Gazebo version 5 or newer?

[image]

durgesh0071, 2017-07-10:
Yes ,I followed those instructions for setting up the virtual joystick.
I am using Gazebo version 7.

closed 2017-07-10 by durgesh0071

endlesswho, 2018-01-10:
I have the same question you meet, have you solved this problem?

lucasmartins131, 2022-07-22:
Guys ... what is the solution for this issue ?