ethz-asl/rotors_simulator #421: mav_with_keyboard:  ERROR: cannot launch node of type [joy/joy_node]: joy

ChenYunan, 2017-07-10:
When I run **Usage with a keyboard**, 
`roslaunch rotors_gazebo mav_with_keyboard.launch mav_name:=firefly world_name:=basic`,
there is an error: **cannot launch node of type [joy/joy_node]:joy.**
I have the Virtual RC joystick interface, and a Gazebo with the firefly, however it can't fly and can't be manipulated neither.
[image]

Is anyone can tell me what's going on and how to solve this problem?
Thanks a lot!

pvechersky, 2017-07-11:
Hi @ChenYunan,

It looks like you're running on ROS kinetic and I had a similar issue with kinetic where the 'joy' package was not installed by default. Try the following command and see if that fixes the issue:

`sudo apt-get install ros-kinetic-joy`

In the meantime I will also update the Wiki page with some of this info. Thank you!

ChenYunan, 2017-07-13:
@pvechersky Thank you very much!  I'm indeed running on ROS kinetic, this problem has been solved. 

BTW, I think the first command of Basic Usage 
`roslaunch rotors_gazebo mav_hovering_example.launch mav_name:=firefly world_name:=basic`
should be 
`roslaunch rotors_gazebo mav.launch`.

pvechersky, 2017-07-14:
@ChenYunan Glad to hear it's working now! :) Can the issue be closed?

Regarding the basic usage command - I think one of the admins would be better to ask about this and in a separate thread, but my guess is that while 'mav.launch' is indeed the most basic of the available launch files, it is 'mav_hovering_example.launch' that actually starts to demonstrate the working functionality of RotorS in action, rather than simply loading a world and a model.

closed 2017-08-02 by pvechersky

Marslanali, 2020-09-13:
Hi,

I am using the same example with ROS Kinetic & Ubuntu 16.04 and I've installed `sudo apt-get install ros-kinetic-joy`, however, after I run `roslaunch rotors_gazebo mav_hovering_example.launch mav_name:=firefly world_name:=basic`. I keep getting an error on my terminal. 

```
[ERROR] [1600017760.234552256]: Couldn't open joystick force feedback!
[firefly/rotors_joy_interface-7] process has died [pid 5534, exit code -11, cmd /home/arslan/catkin_ws/devel/lib/rotors_joy_interface/rotors_joy_interface __name:=rotors_joy_interface __log:=/home/arslan/.ros/log/b897ba14-f5e5-11ea-b649-90324b6bccf7/firefly-rotors_joy_interface-7.log].
log file: /home/arslan/.ros/log/b897ba14-f5e5-11ea-b649-90324b6bccf7/firefly-rotors_joy_interface-7*.log
```

[image]

The Virtual RC joystick interface shows and up and works with the keyboard, however, I can not fly MAV using the keyboard. 
Also, I've no external joystick attached and js is selected to  `/dev/input/js0` 

Any comments on this issue? 

Thanks & looking forward.

Marslanali, 2020-09-14:
Hi, 

I am able to solve the problem, the workaround I had used is adding a  `args=" _dev_name:=python-uinput"` in `mav_with_keyboard.launch` file located in `~/catkin_ws/src/rotors_simulator/rotors_gazebo/launch/`.

```
   <!-- <node name="key_joystick" pkg="rotors_joy_interface" type="key_joystick.py" />  -->
    <node name="joy_node" pkg="joy" type="joy_node" args=" _dev_name:=python-uinput" />
    <node name="rotors_joy_interface" pkg="rotors_joy_interface" type="rotors_joy_interface" />
```

However, before running `mav_with_keyboard.launch` I had to run `rosrun rotors_joy_interface key_joystick.py` so that `python-uinput` at `/dev/input/js1` is available.

[image]

Tcloser, 2020-10-15:
> Hi,
> 
> I am able to solve the problem, the workaround I had used is adding a `args=" _dev_name:=python-uinput"` in `mav_with_keyboard.launch` file located in `~/catkin_ws/src/rotors_simulator/rotors_gazebo/launch/`.
> 
> ```
>    
>     <node name="joy_node" pkg="joy" type="joy_node" args=" _dev_name:=python-uinput" />
>     <node name="rotors_joy_interface" pkg="rotors_joy_interface" type="rotors_joy_interface" />
> ```
> 
> However, before running `mav_with_keyboard.launch` I had to run `rosrun rotors_joy_interface key_joystick.py` so that `python-uinput` at `/dev/input/js1` is available.
> 
> [image]

Open a new issue then someone will reply to you.