ethz-asl/rotors_simulator #423: Increase Odometry sensor rate

JD-ETH, 2017-07-28:
I am working adding my own quadcopter design into the rotors simulator. My onboard controller runs at 1kHz, but I found out the odometry ground truth publishes in a significantly lower rate that make my quadcopter unstable.
 In my case I am pushing the gains to it's physical limits, so the delay is very crucial.
Is there a way to publish the odometry more often in the plugin?

fdiewald, 2017-07-30:
Im assuming you are using the odometry plugin? From my understanding the odometry plugin already publishes a message for each simulation step of the gazebo physics solver if the measurement_divisor is set to 1 (which is the default).

At what rate is your simulation running? At a max step size of 0.001 we should expect to see the message published at ~1kHz assuming gazebo is running at a real time update rate and the real time factor is 1.0. Depending on the load and scene complexity this factor might drop because the physics solver cannot keep up.

So my advice would be to check the settings of the physics solver and to make sure the solver can keep up with real time. If the solver cannot keep up it might be possible to account for the reduced simulation rate by using the simulation time in ros.

JD-ETH, 2017-07-31:
@fdiewald Thanks for the suggestion. Indeed the max step size is set up to 2 ms or 10 ms in most worlds that was shipped with rotors. After changing it to 1ms and real_time_update_rate to 1000 the result seems to improve. 

My solver couldn't keep up and have a real time factor smaller than 1. (in extrem case of physical contact even down to 0.1). What do you mean by use the simulation time in ros to account for this?

fdiewald, 2017-07-31:
It is possible to instruct gazebo via the argument
`<arg name="use_sim_time" value="true"/>`
to publish its time data to the /clock ros topic and to set the "/use_sim_time" parameter on the ros parameter server. The ROS Time Api in your nodes will now listen to this topic for timing information (how long to sleep nodes to reach target rate etc.) instead of your computers wall clock. Check e.g. [Clock](http://wiki.ros.org/Clock) for reference.

Please note that this only applies to the ROS Time Api and not boost/asio timers for instance. You would have to make sure that ALL your components rely entirely on the ROS Time API for time/timing information.

JD-ETH, 2017-08-03:
After setting up the use_sim_time parameter and changing the world file, the odometry call back in my controller is still not receiving odometry information from the plugin at the desired rate:

>   odometry_sub_ = nh.subscribe(kDefaultOdometryTopic, 1,
>                                &QuaternionControllerNode::OdometryCallback, this);

The odometry is  being called back only in some intervals:

> Odometry read At current time 20.804
> Odometry read At current time 20.804
> Odometry read At current time 20.838
> Odometry read At current time 20.838
> Odometry read At current time 20.838
> Odometry read At current time 20.839
> Odometry read At current time 20.872
> Odometry read At current time 20.905
> 

Is there a way to ensure the callback being called in every 1ms? maybe with a timer? I am still unsure where the problem is. Attached the launch and world file.

`<launch>
  <arg name="mav_name" default="felios"/>
  <arg name="world_name" default="empty"/>
  <arg name="enable_logging" default="false" />
  <arg name="enable_ground_truth" default="true" />
  <arg name="log_file" default="$(arg mav_name)" />
  <arg name="use_sim_time" value="true" />
  <env name="GAZEBO_MODEL_PATH" value="${GAZEBO_MODEL_PATH}:$(find rotors_gazebo)/models"/>
  <env name="GAZEBO_RESOURCE_PATH" value="${GAZEBO_RESOURCE_PATH}:$(find rotors_gazebo)/models"/>
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find rotors_gazebo)/worlds/$(arg world_name).world"/>
    
    <arg name="paused" value="true"/>
    
  </include>

  <group ns="$(arg mav_name)">
    <include file="$(find rotors_gazebo)/launch/spawn_mav.launch">
      <arg name="mav_name" value="$(arg mav_name)" />
      <arg name="model" value="$(find rotors_description)/urdf/mav_generic_odometry_sensor.gazebo" />
      <arg name="enable_logging" value="$(arg enable_logging)" />
      <arg name="enable_ground_truth" value="$(arg enable_ground_truth)" />
      <arg name="log_file" value="$(arg log_file)"/>
    </include>
    <node name="joy_node" pkg="joy" type="joy_node" >
      <param name="is_velocity_control" value="false" />
    </node>
    <node name="rotors_joy_interface" pkg="rotors_joy_interface" type="rotors_joy_interface" />
    <node name="quaternion_controller_node" pkg="rotors_control" type="quaternion_controller_node" output="screen">
      <rosparam
        command="load" file="$(find rotors_gazebo)/resource/quaternion_controller_$(arg mav_name).yaml" />
      <rosparam command="load" file="$(find rotors_gazebo)/resource/$(arg mav_name).yaml" />
      <param name="enable_inverted_motor" value="false" />
      <remap from="odometry" to="odometry_sensor1/odometry" />
    </node>
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
  </group>
</launch>
`
`<?xml version="1.0" ?>
<sdf version="1.5">
  <world name="default">
    
    <include>
      <uri>model://sun</uri>
    </include>
    
    
    <plugin name="ros_interface_plugin" filename="librotors_gazebo_ros_interface_plugin.so"></plugin>

    <include>
      <uri>model://ground_plane</uri>
    </include>
    <include>
      <uri>model://asphalt_plane</uri>
    </include>
    <physics name='default_physics' default='0' type='ode'>
      <gravity>0 0 -9.8066</gravity>
      <ode>
        <solver>
          <type>quick</type>
          <iters>10</iters>
          <sor>1.3</sor>
          <use_dynamic_moi_rescaling>0</use_dynamic_moi_rescaling>
        </solver>
        <constraints>
          <cfm>0</cfm>
          <erp>0.2</erp>
          <contact_max_correcting_vel>100</contact_max_correcting_vel>
          <contact_surface_layer>0.001</contact_surface_layer>
        </constraints>
      </ode>
      <max_step_size>0.001</max_step_size>
      <real_time_factor>1</real_time_factor>
      
      <real_time_update_rate>1000</real_time_update_rate>
      <magnetic_field>6e-06 2.3e-05 -4.2e-05</magnetic_field>
    </physics>
  </world>
</sdf>
`

fdiewald, 2017-08-03:
Im afraid i wont be able to help with this competently as i have no experience dealing with this kind of issue but maybe i can give you a few ideas/starting points in passing:

- some users saw improvement with this issue when using the noTcpDelay Transport hint for their subscribers (see [here](https://github.com/ros/ros_comm/issues/974)) this user specifically mentions odometry topics in this context as well.
- at what rate is your controller node (or whoever is the subscriber) running? [here](http://answers.ros.org/question/53055/ros-callbacks-threads-and-spinning/) is a small post about ros communication explaining the theory behind the callbackqueue and spinners. if you are not processing your callback queue at 1khz this could explain the interval in your callbacks (looking like ~33ms or 30Hz from your snippet)
- check your topic for dropped messages (see how to [here](http://wiki.ros.org/Topics#Topic_statistics))
- how complex is your callback? remember to keep your callback very simple at such high rates.

best of luck

JD-ETH, 2017-08-03:
Thanks @fdiewald ! The noTcpDelay option indeed solved the issue. My quad finally makes a stable flight :) 

Looks like there is still quite some magic going on in the communication, I will have a look at the links you posted.

closed 2017-08-03 by JD-ETH

David-Willo, 2019-04-16:
I am also the using the odometry plugin, but the framerate of the odometry topic drops after 0.4s every time I record a rosbag, anyone know the reason? 
[image]