ethz-asl/rotors_simulator #424: Quad becomes uncontrollable with virtual joystick

garg-akash, 2017-08-02:
Whenever I execute roslaunch rotors_gazebo mav_with_keyboard.launch and provide throttle with the W and S, the quadcopter becomes uncontrollable and flies high immediately. What to do?

pvechersky, 2017-08-02:
Hi @garg-akash,

One thing you can do is change the value of the 'max_thrust' parameter for the joy_interface node (you can pass it via the launch file). This value will depend on the MAV model, but I think that the current default value is a bit too high for the default Firefly model.

Lowering this number should make the throttle less sensitive and produce more reasonable forces. Let me know if this works for you.

Torgenn, 2018-02-02:
I was having an issue where the thrust axis was reversed (i.e. the quadcopter would fly away as soon as I unpaused the simulation with the throttle at minimum) and changing the parameter axis_direction_thrust was not working properly. I had to change the cpp file in rotors_joy_interface/src. It's a single line change:
`control_msg_.thrust.z = (msg->axes[axes_.thrust] * axes_.thrust_direction + 1) * max_.thrust / 2.0 ;`
Instead of `control_msg_.thrust.z = (msg->axes[axes_.thrust] + 1) * max_.thrust / 2.0 * axes_.thrust_direction;`
However the quadcopter remains really hard to control even after reducing the max_thrust parameter to 20

marco-marino95, 2020-02-22:
Io non riesco a controllare il quad. Ho seguito questa guida "Imposta il joystick della tastiera virtuale" pero' quando arrivo al punto "rosrun rotors_joy_interface key_joystick.py" non esce la GUI joystick della tastiera virtuale ma mi da questo ERRORE : 

[rosrun] Couldn't find executable named key_joystick.py below /home/marco/catkin_ws/src/CrazyS/rotors_joy_interface
[rosrun] Found the following, but they're either not files,
[rosrun] or not executable:
[rosrun]   /home/marco/catkin_ws/src/CrazyS/rotors_joy_interface/scripts/key_joystick.py

Cosa posso fare per risolvere il problema?

gsilano, 2020-02-22:
Hi @marco-marino95!! I think no one will reply to your message since you wrote it in Italian ;-)
Please, use the EDIT feature to replace the Italian text with the English one

tooHotSpot, 2020-04-09:
@colineRamee Thanks!
Unfortunately, also nothing helped to make movement slower and consistent, even after rebuilding the library. Drone flies up away from the origin. If it was needed, rebuilding the library also did not help.