ethz-asl/rotors_simulator #426: Underlying dynamics that causes attitude drift?

JD-ETH, 2017-08-09:
When using a quaternion controller following Emil Fresk and George Nikolakopoulos "http://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=6669617",  I notice that simply using a PP controller was not enough to prevent the attitude from drifting after a step response.  

Observed behavior:  Motor pairs are spinning at different rpms correctly trying to compensate for the error in attitude (roll/pitch), but the angular velocity does not seem to change and the attitude slowly drifts further away. 

Where are all the forces and torque applied in the simulation? Are they all applied in the motor plugin? Is there some dynamics here that I have neglected?

JD-ETH, 2017-08-09:
After turning off the "rolling moment" in the simulation this issue appears to be solved. 

I don't really understand what this is based on. But this drifting behavior doesn't appear on the real setup we have, so personally I doubt correctness of the modelling.

Maybe someone can post a paper about this?

fdiewald, 2017-08-16:
hey jd,
to my knowledge the only place forces are applied by rotors_simulator is the motor_plugin and the wind plugin should you use it (not the default).

what are you referencing when you are saying to turn off "air drag". are you talking about the 'rotor_drag_coefficient' and 'rolling_moment_coefficient' constants in the quadrocopter descriptions? if so: the paper cited in the source code for the dynamics is [Philippe Martin, Erwan Salaun. The True Role of Accelerometer Feedback in Quadrotor Control. 2009](https://hal.archives-ouvertes.fr/hal-00422423v1/document).

JD-ETH, 2017-08-18:
yes this is exactly what I am looking for. Thanks!

closed 2017-08-18 by JD-ETH