ethz-asl/rotors_simulator #428: error when building this tool

peilibuaa, 2017-08-24:
when build the rotors_simulator with the command 'catkin build', the following errors shows up
any one could give me some guidance on what can i do to solve the problem?
thank you  very much
`_______________________________________________________________________________
Errors     << rotors_gazebo_plugins:make 
/home/lipei/rotors_ws/logs/rotors_gazebo_plugins/build.make.001.log
make[2]: PROTOBUF_PROTOC_EXECUTABLE-NOTFOUND: Command not found
make[2]: *** [Odometry.pb.cc] Error 127
make[1]: *** [CMakeFiles/mav_msgs.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/lipei/rotors_ws/build/rotors_gazebo_plugins; catkin build --get-env rotors_gazebo_plugins | 
catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

...............................................................................
Failed     << rotors_gazebo_plugins:make           [ Exited with code 2 ]      
Failed    <<< rotors_gazebo_plugins                [ 0.2 seconds ]             
Abandoned <<< rotors_gazebo                        [ Unrelated job failed ]    
`
[image]

helenol, 2017-08-24:
Try this out:
`sudo apt-get install protobuf-compiler libprotoc-dev`

Michael9792, 2017-08-24:
Yeah， I believe @peilibuaa you could refer to #352 first.

ffurrer, 2017-08-24:
@peilibuaa if this solved your problem, please close this. Cheers

peilibuaa, 2017-08-24:
thank you very much. it works

closed 2017-08-24 by peilibuaa