ethz-asl/rotors_simulator #434: Protobuf error in Ubuntu 16.04 and libprotoc 3.1.0

ToniRV, 2017-09-21:
Self-note, but might be helpful for others...

I solved this issue a long time ago, but if I remember correctly:
The error I had was along the lines of: "compiled with a newer version than..."

The issue was solved by using https://github.com/ethz-asl/protobuf_catkin instead of the libprotoc I already had.

closed 2017-09-21 by ToniRV

ToniRV, 2017-09-25:
And also try to build rotors_gazebo_plugins alone, voire:
```catkin build rotors_gazebo_plugins --no-deps```

juanmed, 2019-05-27:
Hello @ToniRV , how do you go about using  [ https://github.com/ethz-asl/protobuf_catkin ](https://github.com/ethz-asl/protobuf_catkin)? I am getting an error like this:

```bash
In file included from /home/fer/git_clone/rotors_ws/build/rotors_gazebo_plugins/Actuators.pb.cc:4:0:
/home/fer/git_clone/rotors_ws/build/rotors_gazebo_plugins/Actuators.pb.h:10:40: fatal error: google/protobuf/port_def.inc: No such file or directory
```

I am not very familiar with the cmake files changes required so some guidance would be appreciated. Thanks!