ethz-asl/rotors_simulator #451: Problem installing RotorS on Distro Kinetic

robrother, 2017-11-19:
Hello 

My name is Erick, I´m new using ROS and I´m trying to install RotorS Simulator, mi distro is Kinetic, and when in my workspace I write catkin_make the system says that the package does not folllow version conventions.
 
[image]

Could anybody help me saying me what I need change? 

Greetings from México

Erick

helenol, 2017-11-20:
Try `catkin build rotors_simulator` instead :)

robrother, 2017-11-20:
Hi @helenol , thanks for your soon answer, unfortunately I have the same problem with the version package
[image]
I think that I need change something here, in the version package, but I don't know the correct convention.
[image]

helenol, 2017-11-20:
```
cd ~/catkin_ws/
catkin clean --all
catkin build rotors_simulator
```

Alternatively just delete `~/catkin_ws/build` (the build folder that the first message refers to), but if `catkin clean --all` works this is a nicer option.

robrother, 2017-11-22:
Hi @helenol thanks for your help, finally I could install it, I have to create a new workspace and I do all the steps again.
[image]

closed 2021-01-26 by Jaeyoung-Lim