ethz-asl/rotors_simulator #456: The kinematic Model of pelican

zhuzd12, 2017-12-17:
Guys, I am trying to simulate the LEE controller with pelican quadrotor  in Rotors. But it turns out that when the target point is too far from the origin (about 10m or more), the simulation can not converge to a stable state, namely the target point. It seems that there is something wrong with my code about the kinematic model of pelican, or it could be that I did not take into count the motor model. Can anyone provide the right version of pelican kinematic model or give me some advice?

rikba, 2017-12-17:
Hi @zhuzd12,
this is expected behavior of the Lee controller.

zhuzd12, 2017-12-18:
I found what was wrong with my code finally.  Velocity in Rotors_control::EigenOdometry struct is represent in body frame but not world frame. The Lee controller works well after I corrected it. Thanks for attention.

closed 2017-12-18 by zhuzd12

rikba, 2017-12-18:
Great! What might still happen is that the UAV crashes into the floor if you give set points far away (10m). This is because the position error term in the desired acceleration overshadows the gravitational compensation. See (15) in [Lee, Taeyoung, Melvin Leoky, and N. Harris McClamroch. "Geometric tracking control of a quadrotor UAV on SE (3)." Decision and Control (CDC), 2010 49th IEEE Conference on. IEEE, 2010.](https://pdfs.semanticscholar.org/2e83/b6f1d6da2694dd029597911599c03b690afc.pdf).