ethz-asl/rotors_simulator #458: Catkin Build Error [mavlink]

aka-LJ, 2018-01-02:
Hi,

I am on Ubuntu 16.04 with ROS Kinetic, I got the error message after the catkin build step.
any ideas? 

lj@Ubuntu:~/catkin_ws/src$ cd ~/catkin_ws/
lj@Ubuntu:~/catkin_ws$ catkin build
-------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /home/lj/catkin_ws
-------------------------------------------------------
Source Space:       [exists] /home/lj/catkin_ws/src
Log Space:         [missing] /home/lj/catkin_ws/logs
Build Space:        [exists] /home/lj/catkin_ws/build
Devel Space:        [exists] /home/lj/catkin_ws/devel
Install Space:      [unused] /home/lj/catkin_ws/install
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------
[build] Found '19' packages in 0.0 seconds.                                                                                                                          
[build] Updating package table.                                                                                                                                      
Starting  >>> catkin_tools_prebuild                                                                                                                                  
Finished  <<< catkin_tools_prebuild                [ 1.8 seconds ]                                                                                                   
Starting  >>> mav_msgs                                                                                                                                               
Starting  >>> mavlink                                                                                                                                                
Starting  >>> mavros_msgs                                                                                                                                            
Starting  >>> rotors_comm                                                                                                                                            
Starting  >>> rotors_description                                                                                                                                     
Starting  >>> rotors_evaluation                                                                                                                                      
_____________________________________________________________________________________________________________________________________________________________________
Errors     << mavlink:make /home/lj/catkin_ws/logs/mavlink/build.make.000.log                                                                                        
Traceback (most recent call last):
  File "/home/lj/catkin_ws/src/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/lj/catkin_ws/src/mavlink/pymavlink/generator/mavgen.py", line 12, in <module>
    from future import standard_library
ImportError: No module named future
Traceback (most recent call last):
  File "/home/lj/catkin_ws/src/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/lj/catkin_ws/src/mavlink/pymavlink/generator/mavgen.py", line 12, in <module>
    from future import standard_library
ImportError: No module named future
make[2]: *** [include/v1.0/standard/standard.h] Error 1
make[1]: *** [CMakeFiles/standard.xml-v1.0.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [include/v1.0/ardupilotmega/ardupilotmega.h] Error 1
Traceback (most recent call last):
  File "/home/lj/catkin_ws/src/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/lj/catkin_ws/src/mavlink/pymavlink/generator/mavgen.py", line 12, in <module>
    from future import standard_library
ImportError: No module named future
make[1]: *** [CMakeFiles/ardupilotmega.xml-v1.0.dir/all] Error 2
[... 128 lines cut ...]
bash: syntax error near unexpected token `>'
lj@Ubuntu:~/catkin_ws$ Finished  <<< rotors_control                       [ 25.0 seconds ]                                                                              
Finished: command not found
lj@Ubuntu:~/catkin_ws$ Starting  >>> rotors_gazebo_plugins                                                                                                              
bash: syntax error near unexpected token `>'
lj@Ubuntu:~/catkin_ws$ _________________________________________________________________________________________________________________________________________________
_________________________________________________________________________________________________________________________________________________: command not found
lj@Ubuntu:~/catkin_ws$ Errors     << rotors_gazebo_plugins:make /home/patrik/catkin_ws/logs/rotors_gazebo_plugins/build.make.000.log                                    
> make[2]: PROTOBUF_PROTOC_EXECUTABLE-NOTFOUND: Command not found
> make[2]: *** [Wrench.pb.cc] Error 127
> make[1]: *** [CMakeFiles/mav_msgs.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....
> make: *** [all] Error 2
> cd /home/patrik/catkin_ws/build/rotors_gazebo_plugins; catkin build --get-env rotors_gazebo_plugins | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
> .................................................................................................................................................
> Failed     << rotors_gazebo_plugins:make           [ Exited with code 2 ]                                                                        
> Failed    <<< rotors_gazebo_plugins                [ 4.2 seconds ]                                                                               
> Abandoned <<< rotors_gazebo                        [ Unrelated job failed ]                                                                      
> [build] Summary: 3 of 5 packages succeeded.                                                                                                      
> [build]   Ignored:   40 packages were skipped or are blacklisted.                                                                                
> [build]   Warnings:  None.                                                                                                                       
> [build]   Abandoned: 1 packages were abandoned.                                                                                                  
> [build]   Failed:    1 packages failed.                                                                                                          
> [build] Runtime: 34.7 seconds total.                                                                                                             
> [build] Note: Workspace packages have changed, please re-source setup files to use them.^C

helenol, 2018-01-03:
Resolution from here: https://github.com/ArduPilot/pymavlink/issues/25

```
pip install future
```

And if you don't have pip...
https://pip.pypa.io/en/stable/installing/

Hope this helps!

aka-LJ, 2018-01-03:
It worked, thank you.

Hjayant, 2018-12-04:
Hi ,
I am having catkin build problem.And I am using ubuntu16.04 and ros kinetic. Any idea??
[image]
[image]
[image]
[image]

pinkneonsign, 2020-11-25:
https://github.com/ArduPilot/pymavlink/issues/25#issuecomment-483651986