ethz-asl/rotors_simulator #459: catkin build error

xinhaixufan, 2018-01-09:
I am on Ubuntu16.04 ROS Kinetic. When I run `catkin build `, I got these errors, Thanks!
[image]
[image]
[image]

ffurrer, 2018-01-09:
you need to install octomap_msgs:
```
sudo apt-get install ros-<version>-octomap-msgs
```
or just
```
sudo apt-get install ros-<version>-octomap
```
[octomap_msgs](https://github.com/OctoMap/octomap_msgs)
[octomap](http://wiki.ros.org/octomap)

please close this if it solved your issue

xinhaixufan, 2018-01-20:
Thank you very much, It worked

closed 2018-01-20 by xinhaixufan