ethz-asl/rotors_simulator #469: Documentation Regarding roll ,pitch,yaw rate and thrust controller

chrismath, 2018-03-03:
I wanted to know more information regarding the roll ,pitch,yaw rate and thrust controller i was trying to develop a position controller with it but its not working

gsilano, 2018-03-03:
Dear Chrismath,

you can find more informationi about RotorS in these two chapters book (linked below).

- https://link.springer.com/chapter/10.1007%2F978-3-319-26054-9_23

- https://link.springer.com/chapter/10.1007/978-3-319-54927-9_1

chrismath, 2018-03-03:
Thank you for sharing me these references

Thanking You
Regards
Chris Mathews.N

chrismath, 2018-03-05:
Could also provide the reference for the "lee positional controller". I actually I wanted to know how the command acceleration work.

juanmed, 2019-01-02:
I think that controller is based on https://ieeexplore.ieee.org/document/5717652

gsilano, 2019-01-02:
I suggest to read also https://arxiv.org/abs/1003.2005