ethz-asl/rotors_simulator #470: catkin build;errors rotors_hil_interface:make;mavlink_hil_gps_t

1184, 2018-03-05:
Hi,

I am on Ubuntu 16.04 with ROS Kinetic, I got the error message after the catkin build .Would you help me?

![Uploading 2018-03-05 17-
[image]
13-56屏幕截图.png…]()
[image]
[image]

dannyjose, 2018-03-12:
I had ptoblems too, try 
"pip install future " before "catkin buil"

1184, 2018-03-13:
It woks.Thank you so much.

closed 2018-03-13 by 1184

HusseinHazem, 2018-04-27:
i tried it but rotors_hil_interface still won't build
[image]

v01d, 2018-05-23:
@dannyjose Why should that fix this problem?
@HusseinHazem how did you manage to solve the original build error?

Pedro-Roque, 2018-10-21:
This will fix it:
https://github.com/ethz-asl/rotors_simulator/pull/522/files?fbclid=IwAR2SxViGule-u8S4A1kE0L6HCwW3cixtO7IgMBNTp9TILT62TGdXPlsxApw

Big thumbs up for @TSC21 !

EdXian, 2019-01-14:
Hi @Pedro-Roque ,
I used the ros-melodic version and I tried to build rotors_simulation. I got the same error which @1184  had described. I followed the step to fix the problem. However, I got a new error as the follows:
[image]
Can you give me some suggestions?
Thank you

Pedro-Roque, 2019-01-15:
I'm not sure if Melodic is supported. However, it seems that a library is missing and so the namespace math is not being recognized.

Pedro-Roque, 2019-01-17:
@EdXian please read
https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo/launch/firefly_swarm_hovering_example.launch

and other RotorS doc