ethz-asl/rotors_simulator #472: Mav with Keyboard Launch

weemanlittle, 2018-03-15:
Hello! I have set up the RotorS simulator in an attempt to run the mav_with_keyboard.launch. I have followed the tutorials to set up RotorS and to download python-uinput. The Virtual RC Joystick works and reacts with the keyboard inputs but the drone does not fly, any possible fixes?  

[image]

vkpankov, 2018-04-01:
@weemanlittle, Hello! I have the same problem.

saikrn112, 2018-04-02:
Did you unpause the gazebo simulation?

vkpankov, 2018-04-02:
It works now after remapping device and topic /joy->/firefly/joy
But immediately after launch it flies in one direction, and it is unclear how to control it.

weemanlittle, 2018-04-03:
@saikrn112 Yes, the physics has been unpaused using the rosservice call. As you can see in the image the command worked but the drone does not react to the keyboard inputs. I have not tried what @vkpankov has done but it seems like that did not fix it.

qsnld, 2018-04-18:
@vkpankov ,hello! I have the same problem. I can't control it. It just circles. Did you fix it?

XuanTr, 2018-11-07:
Hello, i have the same problem. Digged into the source code, i find that joystick event (pkg: rotors_joystick_interface) is not trigger. I don't know why. I try to fix without success.

I can run the rotors_joystick_interface joy_node as separated node and publish manually message Joy, it trigger joy event. But when i try it using launch file, it stop to fire the event.

Maybe it's a problem of ros but can not trigger callback.
Anyone have other suggestions ?

XuanTr, 2018-11-08:
I see the problem here, Joystick opened is js0, but not is key-joystick, for my case i add this line to group ns.

<rosparam param="joy_node/dev">/dev/input/js4</rosparam>

Add this line to launch file, it working now.

paul218, 2019-04-08:
> It works now after remapping device and topic /joy->/firefly/joy
> But immediately after launch mav flies in one direction, and it is unclear how to control it.

Hi , i begin witht rotors, i have the same problem, how can you resolve the problem ???

tooHotSpot, 2020-04-09:
@XuanTr Thanks, launched the drone, but with js1 number

ayhangultekin, 2021-02-28:
@tooHotSpot @XuanTr  i have same problem there is no jo file dev/input

and which launch file must i add this <rosparam param="joy_node/dev">/dev/input/js4</rosparam> line

thank you