ethz-asl/rotors_simulator #474: RotorS not working with Gazebo 9

shravank1996, 2018-03-29:
If I build rotors with gazebo9 I am getting the errors of UUID not found and that gazebo.hh isn't found. 

Is there a reason why this happens

vkotaru, 2018-04-15:
same issue

ffurrer, 2018-04-15:
yup, they are not compatible yet, I will try to get it to work when I have some free time, or if someone has time to do it, we are happy to see PRs :).

v01d, 2018-06-27:
The UUID problem can be worked around by adding "ign_import_target(UUID)" right after the find_package for gazebo. However, I still get many issues with gazebo plugins. Anyone worked on this?

ZacharyTaylor, 2018-06-27:
I quickly hacked together a gazebo 9 compatible branch. See this pr https://github.com/ethz-asl/rotors_simulator/pull/497

tahsinkose, 2018-10-23:
Same issue opened in [here](https://bitbucket.org/ignitionrobotics/ign-cmake/issues/40/target-uuid-uuid-does-not-exist). [Last comment](https://bitbucket.org/ignitionrobotics/ign-cmake/issues/40/target-uuid-uuid-does-not-exist#comment-48203461) resolves the problem. Removing `cmake_modules` from `REQUIRED COMPONENTS` did work in my case.

Jaeyoung-Lim, 2021-01-26:
Closing, since this issue seems stale. current master works with gazebo9 and is verified in CI

closed 2021-01-26 by Jaeyoung-Lim