ethz-asl/rotors_simulator #475: Could not wake up Gazebo

xinhaixufan, 2018-04-05:
Ubuntu 16.04
ROS kinetic

It just worked several times and now it does not work. What should I do to resolve  this problems?
Thank you!
[image]
[image]

ffurrer, 2018-04-05:
@xinhaixufan It is a good practice to state what exactly you are trying to do, what you are running and what fails. From this and also the issue in #476, it is hard for us to know what code you launched and what is not working. So if you want help from the community try to be a bit more descriptive in the issues. And if you close an issue it's also a good practice to state what resolved the issue such that others can benefit from your findings.

juanmed, 2018-11-27:
@xinhaixufan  I just had the same issue. If you were installing RotorS and ROS as a fresh new installation, you might just simply try again. Seems probably at first start Gazebo takes longer to start but after a second time its fine.

BloodLemonS, 2024-12-19:
> @xinhaixufan I just had the same issue. If you were installing RotorS and ROS as a fresh new installation, you might just simply try again. Seems probably at first start Gazebo takes longer to start but after a second time its fine.

I have the same problem, but restarting the launch file doesn't solve it