ethz-asl/rotors_simulator #477: rotors_gazebo_plugins fails to build

HusseinHazem, 2018-04-05:
[image]

marco-tranzatto, 2018-04-06:
Did you follow [these instructions](https://github.com/ethz-asl/rotors_simulator/wiki/Setting-up-the-RotorS-Simulator) (point number 3)?
It looks like you didn't install Rotors dependencies (i.e. `glog_catkin`).

HusseinHazem, 2018-04-06:
Yes i installed all the packages required from the mpc library which are the same as the one you mentioned

aeudes, 2018-04-06:
Not sure if it the same problem but we recently have problem with the ability to find gtest on new install.
It was related to "recent" change on catkin.
You could try to apply this commit: https://github.com/ros/catkin/pull/927/files   on /opt/ros/kinetic/share/catkin/cmake/test/gtest.cmake

HusseinHazem, 2018-04-07:
i am not sure what you are saying tbh aeudes, and i installed the package glog_catkin in the src and even more packages won't build and it itself failed to build

ffurrer, 2018-04-09:
So you did `catkin build glog_catkin` and that succeded or not?
If you did only after installing other packages you probably want to clean your workspace and try re-building everything.
```
catkin clean
catkin build
```

HusseinHazem, 2018-04-09:
I did exactly that

m-erdem, 2018-04-25:
Which gazebo version do you use? As much as I know the package is not compatible with gazebo 9 and 8. I had the same error with version 9, but not quite sure with gazebo8. You may wanna use gazebo7.

HusseinHazem, 2018-04-27:
gazebo 7

HusseinHazem, 2018-04-27:
[image]

helenol, 2018-06-05:
Make sure you have mavros in your workspace:
https://github.com/ethz-asl/mavros
And that it's built. If you still get this error, do ```catkin clean --all``` and rebuild your entire workspace.

CarSanoja, 2018-08-26:
Hello!
I am getting exactly the same error. Somebody solved?

anushmanukyan, 2018-11-15:
Hello, I have a similar issue. I have done all possible steps, but I can not manage to build rotors_gazebo_plugins package. 
I have ubuntu 16.04, gazebo7.
At the end I can launch an example like mav.launch, but then there is no ROS messages that the mav provides, i can see only messages of gazebo.

Anybody could give a hint what could be the issue? 

Thanks in advance.

wuyuzaizai, 2018-12-06:
i got the same error

anushmanukyan, 2018-12-06:
I used docker, and all worked perfectly.